VIKINGYFY/OpenWRT-CI #5: 感谢作者老哥的脚本

rianjskis, 2023-05-01:
请问我的机型是r2s如果想自定义插件选项是否能把我在虚拟机中选好生成的.config文件放到config/Rockchip中并生效我在虚拟机中生成的是6.1内核
[1.txt](https://github.com/VIKINGYFY/OpenWRT-CI/files/11365168/1.txt)
，添加软件源的话请问添加在那个文件，以下是我的config文件以及需要添加的软件源，感谢您，
echo "src-git kenzo https://github.com/kenzok8/openwrt-packages" >> ./feeds.conf.default
echo "src-git small https://github.com/kenzok8/small" >> ./feeds.conf.default
git clone https://github.com/sirpdboy/luci-app-ddns-go.git package/ddns-go
git clone https://github.com/sirpdboy/luci-app-autotimeset package/luci-app-autotimeset
git clone https://github.com/natelol/luci-app-beardropper package/luci-app-beardropper
git clone https://github.com/thinktip/luci-theme-neobird.git package/luci-theme-neobird

VIKINGYFY, 2023-05-02:
按照Plugins.sh中的格式添加插件源即可，不推荐用任何大杂烩源，冲突报错很多

rianjskis, 2023-05-02:
> 按照Plugins.sh中的格式添加插件源即可，不推荐用任何大杂烩源，冲突报错很多

昨天添加了我上面发布的软件源，然后把虚拟机生成的.config文件覆盖到r2s的配置文件里，失败了😄

VIKINGYFY, 2023-05-02:
> > 按照Plugins.sh中的格式添加插件源即可，不推荐用任何大杂烩源，冲突报错很多
> 
> 昨天添加了我上面发布的软件源，然后把虚拟机生成的.config文件覆盖到config/Rockchip文件里，失败了😄

自己认真看我的项目，每个文件都要按照格式添加

rianjskis, 2023-05-08:
> > > 按照Plugins.sh中的格式添加插件源即可，不推荐用任何大杂烩源，冲突报错很多
> > 
> > 
> > 昨天添加了我上面发布的软件源，然后把虚拟机生成的.config文件覆盖到config/Rockchip文件里，失败了😄
> 
> 自己认真看我的项目，每个文件都要按照格式添加

老哥，编译成了，但是请问如果想自定义用户名得话，参考那个文件，感谢。

VIKINGYFY, 2023-05-10:
用户名好像改不了

rianjskis, 2023-05-10:
> 用户名好像改不了

好吧😂😂

closed 2023-05-10 by rianjskis