VIKINGYFY/OpenWRT-CI #10: 你好，我用你的仓库编译的时候出现错误，请问有什么办法解决吗？

zjxxs391, 2023-07-20:
Run dev-drprasad/delete-older-releases@master
  with:
    keep_latest: 6
    delete_tags: true
  env:
    GITHUB_TOKEN: 
    OWRT_TARGET: X64
    OWRT_THEME: design
    OWRT_NAME: OWRT
    OWRT_URL: https://github.com/coolsnowwolf/lede.git
    OWRT_PW: password
    OWRT_IP: 19[2](https://github.com/zjxxs391/openwrt-xiaobao/actions/runs/5603652417/jobs/10250559933#step:15:2).168.10.1
    OWRT_SOURCE: lede.git
    OWRT_DATE: 2[3](https://github.com/zjxxs391/openwrt-xiaobao/actions/runs/5603652417/jobs/10250559933#step:15:3).07.20_0[4](https://github.com/zjxxs391/openwrt-xiaobao/actions/runs/5603652417/jobs/10250559933#step:15:4).17.03
    OWRT_TYPE: X[6](https://github.com/zjxxs391/openwrt-xiaobao/actions/runs/5603652417/jobs/10250559933#step:15:6)4 X[8](https://github.com/zjxxs391/openwrt-xiaobao/actions/runs/5603652417/jobs/10250559933#step:15:8)6
🔴 no GITHUB_TOKEN found. pass `GITHUB_TOKEN` as env

VIKINGYFY, 2023-07-21:
GITHUB_TOKEN没有定义，不知道你咋改的

zjxxs391, 2023-07-21:
我github里有GITHUB_TOKEN，难道是我权限没开齐？需要哪些权限可以说一下吗

VIKINGYFY, 2023-07-22:
自己认真看我的yml文件，lede之类的是ci参数文件，owrt-core是编译核心文件，两个缺一不可

closed 2023-07-25 by VIKINGYFY