VIKINGYFY/OpenWRT-CI #22: 你好,rax3000M插件有问!

mrcosir, 2024-01-18:
使用这个可否单独编译https://github.com/sirpdboy/luci-app-eqosplus这个插件呢?我使用的immortalwrt固件

对于编译没有学习过,所以想问问使用这个项目如何编译?能否有个详细文档?

VIKINGYFY, 2024-01-18:
在Scripts/Packages.sh里加入https://github.com/sirpdboy/luci-app-eqosplus插件源，然后在Config/General.txt里添加插件名CONFIG_PACKAGE_luci-app-eqosplus=y即可

另外，这个似乎是软限速，会对CPU造成极大的负担，想用限速可以去[CloseWRT-CI项目](https://github.com/VIKINGYFY/CloseWRT-CI)里下载闭源驱动的固件，带有硬件限速

mrcosir, 2024-01-18:
> 在Scripts/Packages.sh里加入[luci-app-eqosplus插件源](https://github.com/sirpdboy/luci-app-eqosplus)，然后在Config/General.txt里添加插件名CONFIG_PACKAGE_luci-app-eqosplus=y即可
> 
> 另外，这个似乎是软限速，会对CPU造成极大的负担，想用限速可以去[CloseWRT-CI项目](https://github.com/VIKINGYFY/CloseWRT-CI)里下载闭源驱动的固件，带有硬件限速

非常感谢你的回复,我估计已经使用了很久了.这次是想单独安装一个插件而已.如果要更换固件有点过于折腾.本身IMM自带有限速插件,但是没有定时限速功能.我需要一个时间段来启用它.

mrcosir, 2024-01-18:
能再次麻烦告知一下,在Scripts/Packages.sh什么位置加入https://github.com/sirpdboy/luci-app-eqosplus
[image]
不理解语法

VIKINGYFY, 2024-01-21:
UPDATE_PACKAGE "插件名" "作者/项目" "分支"

closed 2024-01-22 by VIKINGYFY