VIKINGYFY/OpenWRT-CI #23: 编译x86时工作流程能走完，但没有固件包，rockchip正常

love-dy, 2024-01-21:
你好大佬，编译x86时工作流程能走完，但没有固件包，rockchip正常编译有固件包，请问这是什么问题，把所有不必要的插件都去掉也不行，rockchip无论如何都能成功，x86就不行
[image]
[image]
[image]

love-dy, 2024-01-21:
配置如下，其它无改动
[image]
[image]
[image]

love-dy, 2024-01-22:
[image]

VIKINGYFY, 2024-01-22:
我的项目是根据Config/X64.txt里第一行来检索和修改文件的，x86_64固件名里没有包含这个，所以无法打包固件，后续我可能会修改一下

love-dy, 2024-01-22:
我懂了，大佬，非常感谢你的指点，又学到了新知识，你的项目很好用
[image]
[image]

closed 2024-01-22 by VIKINGYFY