VIKINGYFY/OpenWRT-CI #48: QCA-ALL添加Openclash编译失败

kennyhn88, 2024-07-07:
我比较喜欢用openclash
所以就想把它编译进固件
我只是在这里添加了这么一行
就编译出错
请问能改进一下吗
感谢
[image]

dos7lo, 2024-07-07:
很奇怪 只要在下面加openclash或者passwall 就会编译失败 只有第一次 只删除机型的编译成功 后续添加了插件的全失败了 不是shadowsocks-libev就是ruby报错

kennyhn88, 2024-07-07:
这是什么意思“ 只有第一次 只删除机型的编译成功”

dos7lo, 2024-07-07:
`configure: error: MBEDTLS_CIPHER_MODE_CFB required
make[3]: *** [Makefile:130: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocks-libev-3.3.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
checking whether mbedtls supports Cipher Feedback mode or not... make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/net/shadowsocks-libev'
time: package/feeds/packages/shadowsocks-libev/compile#13.93#1.27#15.94
    ERROR: package/feeds/packages/shadowsocks-libev failed to build.
make[2]: *** [package/Makefile:173: package/feeds/packages/shadowsocks-libev/compile] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
make[1]: *** [package/Makefile:167: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/include/toplevel.mk:248: world] Error 2
Error: Process completed with exit code 2.`
报错内容，还有就是自动执行download packages 那块下了一个小时 手动执行就2分钟...

dos7lo, 2024-07-07:
> 这是什么意思“ 只有第一次 只删除机型的编译成功”

第一次没加任何插件 情况和你一样 补充下你的问题

VIKINGYFY, 2024-07-08:
移除了相关依赖的版本更新，应该是没问题了，你们再试试，主要是我不用OC

dos7lo, 2024-07-08:
> 移除了相关依赖的版本更新，应该是没问题了，你们再试试，主要是我不用OC

ok，谢谢，在测试编译，目前没有报错 应该是ok了 还有个疑问想咨询下 开启vlmcsd也会失败 这个能调整吗？
```shell
2024-07-07T23:15:39.5574902Z make[4]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113'
2024-07-07T23:15:39.5645484Z make[5]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/src'
2024-07-07T23:15:39.5749769Z 	CC	vlmcs.o <- vlmcs.c
2024-07-07T23:15:39.5763937Z 	CC	kmsdata-full.o <- kmsdata-full.c
2024-07-07T23:15:39.5781089Z 	CC	crypto.o <- crypto.c
2024-07-07T23:15:39.5804635Z 	CC	kms.o <- kms.c
2024-07-07T23:15:39.5822936Z 	CC	endian.o <- endian.c
2024-07-07T23:15:39.5856878Z 	CC	output.o <- output.c
2024-07-07T23:15:39.5918462Z 	CC	shared_globals.o <- shared_globals.c
2024-07-07T23:15:39.6016075Z 	CC	helpers.o <- helpers.c
2024-07-07T23:15:39.6153254Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6154637Z compilation terminated.
2024-07-07T23:15:39.6176051Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6177224Z compilation terminated.
2024-07-07T23:15:39.6204998Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6206743Z compilation terminated.
2024-07-07T23:15:39.6215210Z make[5]: *** [GNUmakefile:512: ../build/kmsdata-full.o] Error 1
2024-07-07T23:15:39.6216289Z make[5]: *** Waiting for unfinished jobs....
2024-07-07T23:15:39.6217254Z make[5]: *** [GNUmakefile:512: ../build/crypto.o] Error 1
2024-07-07T23:15:39.6218354Z make[5]: *** [GNUmakefile:512: ../build/endian.o] Error 1
2024-07-07T23:15:39.6233872Z 	CC	network.o <- network.c
2024-07-07T23:15:39.6280210Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6281748Z compilation terminated.
2024-07-07T23:15:39.6302163Z make[5]: *** [GNUmakefile:513: ../build/vlmcs.o] Error 1
2024-07-07T23:15:39.6415586Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6417068Z compilation terminated.
2024-07-07T23:15:39.6423296Z make[5]: *** [GNUmakefile:513: ../build/kms.o] Error 1
2024-07-07T23:15:39.6459399Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6461144Z compilation terminated.
2024-07-07T23:15:39.6462029Z make[5]: *** [GNUmakefile:513: ../build/output.o] Error 1
2024-07-07T23:15:39.6504628Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6506023Z compilation terminated.
2024-07-07T23:15:39.6507084Z make[5]: *** [GNUmakefile:513: ../build/shared_globals.o] Error 1
2024-07-07T23:15:39.6678655Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6680227Z compilation terminated.
2024-07-07T23:15:39.6680853Z make[5]: *** [GNUmakefile:513: ../build/helpers.o] Error 1
2024-07-07T23:15:39.6731133Z aarch64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
2024-07-07T23:15:39.6731970Z compilation terminated.
2024-07-07T23:15:39.6733409Z make[5]: *** [GNUmakefile:513: ../build/network.o] Error 1
2024-07-07T23:15:39.6735059Z make[5]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/src'
2024-07-07T23:15:39.6736252Z make[4]: *** [GNUmakefile:111: all] Error 2
2024-07-07T23:15:39.6737475Z make[4]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113'
2024-07-07T23:15:39.6739247Z make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/net/vlmcsd'
2024-07-07T23:15:39.6742295Z make[3]: *** [Makefile:54: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.built] Error 2
2024-07-07T23:15:39.6746886Z time: package/feeds/packages/vlmcsd/compile#0.34#0.18#0.28
2024-07-07T23:15:39.6753337Z     ERROR: package/feeds/packages/vlmcsd failed to build.
2024-07-07T23:15:39.6757156Z make[2]: *** [package/Makefile:173: package/feeds/packages/vlmcsd/compile] Error 1
2024-07-07T23:15:39.6762260Z make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
2024-07-07T23:15:39.6769923Z make[1]: *** [package/Makefile:167: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2024-07-07T23:15:39.6773322Z make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
2024-07-07T23:15:39.6780222Z make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/include/toplevel.mk:248: world] Error 2
2024-07-07T23:15:39.6797055Z ##[error]Process completed with exit code 2.
```

VIKINGYFY, 2024-07-08:
> ok，谢谢，在测试编译，目前没有报错 应该是ok了 还有个疑问想咨询下 开启vlmcsd也会失败 这个能调整吗？
>
> ```shell
> 2024-07-07T23:15:39.5574902Z make[4]: Entering directory [...]

不知道，建议问问gpt，我就简单的用个HP，玩得真没你们花

dos7lo, 2024-07-08:
不好意思 过于乐观了 还是ruby过不了。。。。:(
我添加的软件包：
```shell
#增加插件，科学上网插件请在Scripts/Settings.sh调整
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-app-alist=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-qbittorrent=y
CONFIG_PACKAGE_luci-app-rclone=y
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-smartdns=y
CONFIG_PACKAGE_luci-app-socat=y
``` 
报错内容：
```shell
2024-07-08T05:21:12.0050708Z warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
2024-07-08T05:21:12.0051402Z 
2024-07-08T05:21:12.0926730Z revision.h unchanged
2024-07-08T05:21:12.1363177Z generating x86_64-linux-gnu-fake.rb
2024-07-08T05:21:12.2686105Z x86_64-linux-gnu-fake.rb updated
2024-07-08T05:21:12.2713948Z generating enc.mk
2024-07-08T05:21:12.4760730Z making srcs under enc
2024-07-08T05:21:12.5129749Z make[5]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/hostpkg/ruby-3.3.3'
2024-07-08T05:21:12.5130816Z make[5]: Nothing to be done for 'srcs'.
2024-07-08T05:21:12.5131747Z make[5]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/hostpkg/ruby-3.3.3'
2024-07-08T05:21:12.5145288Z generating transdb.h
2024-07-08T05:21:12.6809705Z transdb.h unchanged
2024-07-08T05:21:12.6893706Z linking static-library libruby-static.a
2024-07-08T05:21:12.8514133Z LLVM ERROR: Invalid encoding
2024-07-08T05:21:12.9504203Z make[4]: *** [Makefile:318: libruby-static.a] Aborted (core dumped)
2024-07-08T05:21:12.9505965Z make[4]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/hostpkg/ruby-3.3.3'
2024-07-08T05:21:12.9507342Z make[4]: *** Deleting file 'libruby-static.a'
2024-07-08T05:21:12.9508500Z make[3]: *** [Makefile:1189: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/hostpkg/ruby-3.3.3/.built] Error 2
2024-07-08T05:21:12.9509478Z make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/lang/ruby'
2024-07-08T05:21:12.9514415Z time: package/feeds/packages/ruby/host-compile#0.88#0.28#1.45
2024-07-08T05:21:12.9522487Z     ERROR: package/feeds/packages/ruby [host] failed to build.
2024-07-08T05:21:12.9526824Z make[2]: *** [package/Makefile:171: package/feeds/packages/ruby/host/compile] Error 1
2024-07-08T05:21:12.9532141Z make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
2024-07-08T05:21:12.9537658Z make[1]: *** [package/Makefile:167: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2024-07-08T05:21:12.9541812Z make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
2024-07-08T05:21:12.9550046Z make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/include/toplevel.mk:248: world] Error 2
2024-07-08T05:21:12.9567856Z ##[error]Process completed with exit code 2.
```

VIKINGYFY, 2024-07-08:
> 不好意思 过于乐观了 还是ruby过不了。。。。:( 我添加的软件包：
>
> ```shell
> #增加插件，科学上网插件请在Scripts/Settings.sh调整
> CONFIG_PACKAGE_luci-app-upnp=y
> CONFIG_PACKAGE_luci-app-diskman=y
> CONFIG_PACKAGE_luci-app-autoreboot=y [...]

ruby报错很久了，我的建议是直接用HP得了

dos7lo, 2024-07-08:
嗯嗯 谢谢大佬 

> ruby报错很久了，我的建议是直接用HP得了

好吧 谢谢大佬 我试试 之前用hp老断。。。。

closed 2024-07-08 by VIKINGYFY