VIKINGYFY/OpenWRT-CI #50: memTotal这个数据差异很大

laomao9000, 2024-07-09:
大侠，比较几个系统，发现memTotal差异很大——
都是256M内存：
[QCA-WIFI-NO_openwrt-6.x.git_main_24.07.09_04.23.56]
[image]

另一个openwrt，也是256M内存

[image]

其它的内存用在了何处，为何有这么大的差异。

laomao9000, 2024-07-09:
往上跟踪了下，是reserved memory，80M，这个参数设置这么大！是不是可以优化？
 Memory: 181560K/262144K available (8256K kernel code, 940K rwdata, 2460K rodata, 1600K init, 307K bss, 80584K reserved, 0K cma-reserved)

King-Of-Knights, 2024-07-09:
NSS需要固件占用

laomao9000, 2024-07-10:
这个大小从何而来？是厂家设定吗？发现多个源的配置大小设定并不一致。

King-Of-Knights, 2024-07-10:
@laomao9000 https://github.com/King-Of-Knights/openwrt-6.x/blob/main/target/linux/qualcommax/patches-6.6/9999-01-add-serveal-support-for-ipq6018.patch#L21

20ssjy2871, 2024-07-18:
经过前面大佬的提醒，我观察到问题大概率出现在这块内存分区 ：https://github.com/King-Of-Knights/openwrt-6.x/blob/b26d71e22c9712e72768d76dc84c9e649e8a1d10/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq6018-256m.dtsi#L10
这部分分区在4.4内核被补丁分为4M大小，到了6.1内核以上就被分到30~50M之间；
似乎可以通过修改其分区的大小来增加可用内存。
我的短暂尝试是，我将其分为4M及以下时系统较为不稳定，频繁重启，当然也不排除与编译库源码有一定关系，，我目前将其改为8~9M的样子，仍然未经长期使用测试。PS，我只有科学上网和upnp插件
在我看来256M内存 基本可以放弃使用带无线功能的op了，除非你为了运行无线放弃使用其它插件

20ssjy2871, 2024-07-18:
经过前面大佬的提醒，我观察到问题大概率出现在这块内存分区 ：
https://github.com/King-Of-Knights/openwrt-6.x/blob/b26d71e22c9712e72768d76dc84c9e649e8a1d10/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom/ipq6018-256m.dtsi#L10
这部分分区在4.4内核被补丁分为4M大小，到了6.1内核以上就被分到30-50M之间；
似乎可以通过修改其分区的大小来增加可用内存。
我的短暂尝试是，我将其分为4M及以下时系统较为不稳定，频繁重启，当然也不排除与编译库源码有一定关系，，我目前将其改为8-9M的样子，仍然未经长期使用测试。PS，我只有科学上网和upnp插件
在我看来256M内存 基本可以放弃使用带无线功能的op了，除非你愿意为了运行无线放弃使用其它插件，另外4.4内核肯定可用内存多一些

closed 2024-08-15 by VIKINGYFY