VIKINGYFY/OpenWRT-CI #51: QCA_All添加openlacsh失败

kennyhn88, 2024-07-09:
openclash是一个很常见的软件
我想把它编译进固件
结果只要添加openclash就编译失败
请问有什么解决的办法吗
感谢
这条我前天发过，是不是隔一天帖子就会被删除呀？
[image]

VIKINGYFY, 2024-07-09:
ruby依赖在6.6内核下报错很久了

kennyhn88, 2024-07-09:
有什么解决办法吗

VIKINGYFY, 2024-07-09:
换个其他的，hp，pw，mihomo

kennyhn88, 2024-07-09:
还有
opkg update报错 
可以怎么解决吗

root@Athena:~# opkg update
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq60xx/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq60xx/packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/libwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/libwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/nss_packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/nss_packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/libwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/libwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/sqm_scripts_nss/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/sqm_scripts_nss/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/libwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq60xx/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/nss_packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/sqm_scripts_nss/Packages.gz, wget returned 8.

VIKINGYFY, 2024-07-09:
挂梯子

kennyhn88, 2024-07-09:
不是梯子问题
[image]

VIKINGYFY, 2024-07-09:
等上游修复吧，玫瑰大佬可能要换成imm源

closed 2024-07-14 by VIKINGYFY