VIKINGYFY/OpenWRT-CI #63: 8月16日编译版本，ZNM2无线突然消失然后自己恢复正常，这是当时的日志

reaichangzai, 2024-08-16:
Fri Aug 16 23:54:38 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED b2:3a:4e:e2:07:25
Fri Aug 16 23:54:38 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 IEEE 802.11: authenticated
Fri Aug 16 23:54:38 2024 kern.warn kernel: [17163.436450] Ignoring NSS change in VHT Operating Mode Notification from b2:3a:4e:e2:07:25 with invalid nss 1
Fri Aug 16 23:54:38 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 IEEE 802.11: associated (aid 1)
Fri Aug 16 23:54:38 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED b2:3a:4e:e2:07:25 auth_alg=open
Fri Aug 16 23:54:38 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 RADIUS: starting accounting session 057DCF955F8572E9
Fri Aug 16 23:54:38 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 WPA: pairwise key handshake completed (RSN)
Fri Aug 16 23:54:38 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED b2:3a:4e:e2:07:25
Fri Aug 16 23:54:38 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.142 b2:3a:4e:e2:07:25
Fri Aug 16 23:54:38 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.142 b2:3a:4e:e2:07:25 MI-8
Fri Aug 16 23:55:01 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED b2:3a:4e:e2:07:25
Fri Aug 16 23:55:03 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 IEEE 802.11: authenticated
Fri Aug 16 23:55:03 2024 kern.warn kernel: [17163.441632] Ignoring NSS change in VHT Operating Mode Notification from b2:3a:4e:e2:07:25 with invalid nss 2
Fri Aug 16 23:55:03 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 IEEE 802.11: associated (aid 1)
Fri Aug 16 23:55:03 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED b2:3a:4e:e2:07:25 auth_alg=open
Fri Aug 16 23:55:03 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 RADIUS: starting accounting session F90059BE9FE36369
Fri Aug 16 23:55:03 2024 daemon.info hostapd: phy0-ap0: STA b2:3a:4e:e2:07:25 WPA: pairwise key handshake completed (RSN)
Fri Aug 16 23:55:03 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED b2:3a:4e:e2:07:25
Fri Aug 16 23:55:03 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.142 b2:3a:4e:e2:07:25
Fri Aug 16 23:55:03 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.142 b2:3a:4e:e2:07:25 MI-8
Fri Aug 16 23:55:28 2024 daemon.err uhttpd[3576]: [info] luci: accepted login on / for root from

closed 2024-08-16 by reaichangzai

master-yun-yun, 2024-08-18:
怎么好的？
舍不得说两句？

reaichangzai, 2024-08-18:
> 怎么好的？ 舍不得说两句？

没有好，我等他再出现一次以后再给大佬提吧