VIKINGYFY/OpenWRT-CI #80: 佬，咨询一下，我用的京东云亚瑟AX1800Pro，编译固件里面加入了docker怎么没刷成功。

mrxie68, 2024-10-05:
佬，咨询一下，我用的京东云亚瑟AX1800Pro，编译固件里面加入了docker，也编译成功了，但是刷机的时候刷不成功。是不支持docker吗？还是我的食用方式有问题呢？
# 启用 Qualcomm IPQ60XX 平台
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq60xx=y

# 仅编译 jdcloud_ax1800-pro 设备
CONFIG_TARGET_DEVICE_qualcommax_ipq60xx_DEVICE_jdcloud_ax1800-pro=y

# Wi-Fi 驱动包（保留 jdcloud_ax1800-pro 的 Wi-Fi 驱动）
CONFIG_TARGET_DEVICE_PACKAGES_qualcommax_ipq60xx_DEVICE_jdcloud_ax1800-pro="ipq-wifi-jdcloud_ax1800-pro"

# 启用 ath11k 相关的 Wi-Fi 驱动模块
CONFIG_PACKAGE_kmod-ath11k=y
CONFIG_PACKAGE_kmod-ath11k-ahb=y
CONFIG_PACKAGE_kmod-ath11k-pci=y
CONFIG_PACKAGE_ath11k-firmware-ipq6018=y
CONFIG_PACKAGE_ath11k-firmware-qcn9074=y

# NSS（Network Subsystem）驱动设置
CONFIG_ATH11K_NSS_SUPPORT=y
CONFIG_ATH11K_NSS_MESH_SUPPORT=y
CONFIG_PACKAGE_MAC80211_NSS_SUPPORT=y

# 其他内核和调试配置
CONFIG_ATH11K_DEBUGFS_HTT_STATS=n
CONFIG_ATH11K_DEBUGFS_STA=n
CONFIG_ATH11K_THERMAL=n
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_KERNEL_ARM_PMU=y
CONFIG_KERNEL_ARM_PMUV3=y
CONFIG_KERNEL_DYNAMIC_DEBUG=y
CONFIG_KERNEL_PERF_EVENTS=y
CONFIG_KERNEL_PREEMPT_NONE_BUILD=y
CONFIG_KERNEL_PREEMPT_NONE=y

# 禁用不必要的包源
CONFIG_FEED_nss_packages=n
CONFIG_FEED_sqm_scripts_nss=n

#插件调整
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-dockerman=y

VIKINGYFY, 2024-10-05:
CONFIG_IPQ_MEM_PROFILE_1024=n
CONFIG_IPQ_MEM_PROFILE_512=y
CONFIG_ATH11K_MEM_PROFILE_1G=n
CONFIG_ATH11K_MEM_PROFILE_512M=y

ATH11K内存配置只能选512，选1G会无法启动

mrxie68, 2024-10-05:
Settings.sh 里面好像有这个判断
#高通平台锁定512M内存
if [[ $WRT_TARGET == *"IPQ"* ]]; then
	echo "CONFIG_IPQ_MEM_PROFILE_1024=n" >> ./.config
	echo "CONFIG_IPQ_MEM_PROFILE_512=y" >> ./.config
	echo "CONFIG_ATH11K_MEM_PROFILE_1G=n" >> ./.config
	echo "CONFIG_ATH11K_MEM_PROFILE_512M=y" >> ./.config
fi

VIKINGYFY, 2024-10-05:
那就不是很清楚了，不玩docker

closed 2024-10-07 by mrxie68