VIKINGYFY/OpenWRT-CI #81: 遇到有关uhttpd的自己目前解决不了的问题

washsky, 2024-10-06:
遇到困境一:uhttpd的配置文件'/etc/config/uhttpd'添加对https端口监听的一行'list listen_https '[::]:443'后,重启路由器添加的这一行就消失了,所以遇到重启路由器这个配置文件就还原了,在/etc/rc.local添加命令也没有用,貌似/etc/rc.local也不可用

遇到困境二(不是大问题):尝试移除'uhttpd'使用nginx代替,但是下面那段配置文件依旧会编译uhttpd在固件里面,不过nginx还是正常使用的,问题不大,:"# 不使用 uhttpd
# CONFIG_PACKAGE_luci-app-uhttpd is not set
# CONFIG_PACKAGE_luci-ssl-openssl is not set
# CONFIG_PACKAGE_uhttpd is not set
# CONFIG_PACKAGE_uhttpd-mod-ubus is not set

# 使用 nginx
CONFIG_PACKAGE_nginx=y
CONFIG_PACKAGE_nginx-ssl=y

# Common nginx modules
CONFIG_PACKAGE_nginx-mod-luci-ssl=y  # LuCI SSL support for Nginx
CONFIG_PACKAGE_nginx-mod-luci=y      # LuCI web interface support
CONFIG_PACKAGE_nginx-mod-ubus=y      # UBUS integration for Nginx
CONFIG_PACKAGE_nginx-mod-stream=y    # Stream module support

# LuCI interface for managing nginx
CONFIG_PACKAGE_luci-app-nginx=y
CONFIG_PACKAGE_luci-nginx=y"

washsky, 2024-10-06:
机型是京东云亚瑟ax1800pro 改内存1g的,没有尝试512内存的

washsky, 2024-10-06:
"# 不使用 uhttpd
# CONFIG_PACKAGE_luci-app-uhttpd is not set
# CONFIG_PACKAGE_luci-ssl-openssl is not set
# CONFIG_PACKAGE_uhttpd is not set
# CONFIG_PACKAGE_uhttpd-mod-ubus is not set

# 使用 nginx
CONFIG_PACKAGE_nginx=y
CONFIG_PACKAGE_nginx-ssl=y

# Common nginx modules
CONFIG_PACKAGE_nginx-mod-luci-ssl=y  # LuCI SSL support for Nginx
CONFIG_PACKAGE_nginx-mod-luci=y      # LuCI web interface support
CONFIG_PACKAGE_nginx-mod-ubus=y      # UBUS integration for Nginx
CONFIG_PACKAGE_nginx-mod-stream=y    # Stream module support

# LuCI interface for managing nginx
CONFIG_PACKAGE_luci-app-nginx=y
CONFIG_PACKAGE_luci-nginx=y"

washsky, 2024-10-06:
"# 不使用 uhttpd"
"# CONFIG_PACKAGE_luci-app-uhttpd is not set"
"# CONFIG_PACKAGE_luci-ssl-openssl is not set"
"# CONFIG_PACKAGE_uhttpd is not set"
"# CONFIG_PACKAGE_uhttpd-mod-ubus is not set"

"# 使用 nginx"
CONFIG_PACKAGE_nginx=y
CONFIG_PACKAGE_nginx-ssl=y

"# Common nginx modules"
CONFIG_PACKAGE_nginx-mod-luci-ssl=y  # LuCI SSL support for Nginx
CONFIG_PACKAGE_nginx-mod-luci=y      # LuCI web interface support
CONFIG_PACKAGE_nginx-mod-ubus=y      # UBUS integration for Nginx
CONFIG_PACKAGE_nginx-mod-stream=y    # Stream module support

"# LuCI interface for managing nginx"
CONFIG_PACKAGE_luci-app-nginx=y
CONFIG_PACKAGE_luci-nginx=y

VIKINGYFY, 2024-10-06:
不太清楚，建议问GPT，我玩得没你花