VIKINGYFY/OpenWRT-CI #96: 亚瑟1104版本5G连不上

jmzwcn, 2024-11-05:
Tue Nov  5 17:56:17 2024 daemon.notice hostapd: phy0-ap0: STA 2e:61:fc:a6:ae:d3 IEEE 802.11: did not acknowledge authentication response

VIKINGYFY, 2024-11-19:
地区请选CN，不要选其他地方

VIKINGYFY, 2024-11-23:
已更换无线固件，再试试

jmzwcn, 2024-11-24:
还是2.4G没问题

Sun Nov 24 13:44:56 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun Nov 24 13:45:05 2024 daemon.err nmbd[4562]: [2024/11/24 13:45:05.592901,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Sun Nov 24 13:45:05 2024 daemon.err nmbd[4562]:   *****
Sun Nov 24 13:45:05 2024 daemon.err nmbd[4562]:
Sun Nov 24 13:45:05 2024 daemon.err nmbd[4562]:   Samba name server OWRT is now a local master browser for workgroup WORKGROUP on subnet 192.168.10.1
Sun Nov 24 13:45:05 2024 daemon.err nmbd[4562]:
Sun Nov 24 13:45:05 2024 daemon.err nmbd[4562]:   *****
Sun Nov 24 13:45:23 2024 daemon.notice hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: did not acknowledge authentication response
Sun Nov 24 13:45:23 2024 daemon.info hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: authenticated
Sun Nov 24 13:45:23 2024 daemon.info hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: associated (aid 1)
Sun Nov 24 13:45:33 2024 daemon.info hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: deauthenticated due to local deauth request
Sun Nov 24 13:45:48 2024 cron.err crond[5749]: time disparity of 1479 minutes detected
Sun Nov 24 13:45:49 2024 daemon.notice hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: did not acknowledge authentication response
Sun Nov 24 13:45:49 2024 daemon.info hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: authenticated
Sun Nov 24 13:46:00 2024 daemon.err hostapd: nl80211: Driver does not support unsolicited broadcast Probe Response frame transmission for phy1-ap0
Sun Nov 24 13:46:12 2024 daemon.info hostapd: phy1-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: authenticated
Sun Nov 24 13:46:12 2024 daemon.info hostapd: phy1-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: associated (aid 1)
Sun Nov 24 13:46:12 2024 daemon.notice hostapd: phy0-ap0: Prune association for 10:b5:88:6e:7a:67
Sun Nov 24 13:46:13 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 10:b5:88:6e:7a:67 auth_alg=open
Sun Nov 24 13:46:13 2024 daemon.info hostapd: phy1-ap0: STA 10:b5:88:6e:7a:67 RADIUS: starting accounting session 13C7822575946B54
Sun Nov 24 13:46:13 2024 daemon.info hostapd: phy1-ap0: STA 10:b5:88:6e:7a:67 WPA: pairwise key handshake completed (RSN)
Sun Nov 24 13:46:13 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 10:b5:88:6e:7a:67
Sun Nov 24 13:46:16 2024 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 10:b5:88:6e:7a:67
Sun Nov 24 13:46:16 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.10.246 10:b5:88:6e:7a:67
Sun Nov 24 13:46:16 2024 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 10:b5:88:6e:7a:67
Sun Nov 24 13:46:16 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.10.246 10:b5:88:6e:7a:67
Sun Nov 24 13:46:17 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.10.246 10:b5:88:6e:7a:67
Sun Nov 24 13:46:17 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.10.246 10:b5:88:6e:7a:67 jmzwcndeAir
Sun Nov 24 13:46:23 2024 daemon.err uhttpd[3266]: [info] luci: accepted login on / for root from 192.168.10.246
Sun Nov 24 13:46:42 2024 daemon.info hostapd: phy0-ap0: STA 10:b5:88:6e:7a:67 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Nov 24 13:47:02 2024 daemon.err hostapd: nl80211: Driver does not support unsolicited broadcast Probe Response frame transmission for phy1-ap0

VIKINGYFY, 2024-11-24:
你什么设备连的

jmzwcn, 2024-11-24:
> 你什么设备连的，不保留配置刷一下看看，WIFI设置不要动

红米手机，mac-m1本 ，都一样

jmzwcn, 2024-11-24:
同一个亚瑟，刷lean的QWRT，连5G就没这问题，很奇怪。

VIKINGYFY, 2024-11-24:
那就不是很清楚了，我这边暂时没发现这个问题

imShire, 2024-11-29:
同样的情况+1，5g 全设备连不上，2.4g没问题，istoreOS和koksaver/OpenWrt360_6.1 都没有这个问题

qqsir-dev, 2024-11-30:
试试加入这些：
CONFIG_PACKAGE_kmod-ath10k=y
CONFIG_PACKAGE_kmod-ath10k-ct=y
CONFIG_PACKAGE_kmod-cfg80211=y
CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_wpad-basic=y
CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_iwinfo=y

qqsir-dev, 2024-11-30:
要支持WiFi7还得加入这个：
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_kmod-regulatory=y

VIKINGYFY, 2024-11-30:
亚瑟又不支持wifi7，加了干啥
。。

closed 2024-12-09 by VIKINGYFY

jmzwcn, 2024-12-10:
解决了吗，没调查清楚就关了？

VIKINGYFY, 2024-12-11:
> 解决了吗，没调查清楚就关了？

我手头已经没有亚瑟了，问了群友，都没遇到过这个情况，暂时没法处理，你可以换其他固件试试