VIKINGYFY/OpenWRT-CI #103: 如何添加其他APP？

qqsir-dev, 2024-11-24:
看了下代码，如何添加openclash或者mihomo?
如何能根据添加的APP自动安装依赖？
多谢

VIKINGYFY, 2024-11-24:
这两个我已经添加了源，在config里添加包名即可

qqsir-dev, 2024-11-24:
哦哦，如果我想加入oaf呢，我试了试好像不行

VIKINGYFY, 2024-11-24:
没有的源需要自己添加，参考相关文件里的格式添加

qqsir-dev, 2024-11-24:
好的，我再试试，谢谢。

qqsir-dev, 2024-11-24:
touch /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/stamp/.boost_installed
echo "boost" >> /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/pkginfo/boost.default.install
make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/libs/boost'
time: package/feeds/packages/boost/compile#20.65#11.52#34.40
make[3]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/utils/cgroupfs-mount'
mkdir -p /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/cgroupfs-mount-2020.06.26~05494281/.pkgdir/cgroupfs-mount/. /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/stamp/.cgroupfs-mount_installed
echo "cgroupfs-mount" >> /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/pkginfo/cgroupfs-mount.default.install
make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/utils/cgroupfs-mount'
time: package/feeds/packages/cgroupfs-mount/compile#0.15#0.05#0.19
make[3]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/net/chinadns-ng'
mkdir -p /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/chinadns-ng-2024.10.14/.pkgdir/chinadns-ng/. /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/root-x86/stamp/.chinadns-ng_installed
echo "chinadns-ng" >> /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/pkginfo/chinadns-ng.default.install
make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/net/chinadns-ng'
time: package/feeds/packages/chinadns-ng/compile#0.15#0.05#0.19
make[3]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/utils/coremark'
rm -f /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/coremark-d5fad6bd094899101a4e5fd53af7298160ced6ab/.built
touch /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/coremark-d5fad6bd094899101a4e5fd53af7298160ced6ab/.built_check
/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/sed -i -e 's|EXE = .exe|EXE =|' /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/coremark-d5fad6bd094899101a4e5fd53af7298160ced6ab/posix/core_portme.mak
mkdir /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/coremark-d5fad6bd094899101a4e5fd53af7298160ced6ab/x86_64
mkdir: cannot create directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/coremark-d5fad6bd094899101a4e5fd53af7298160ced6ab/x86_64': File exists
make[3]: *** [Makefile:98: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-x86_64_musl/coremark-d5fad6bd094899101a4e5fd53af7298160ced6ab/.built] Error 1
make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/feeds/packages/utils/coremark'
time: package/feeds/packages/coremark/compile#0.13#0.06#0.18
    ERROR: package/feeds/packages/coremark failed to build.
make[2]: *** [package/Makefile:185: package/feeds/packages/coremark/compile] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
make[1]: *** [package/Makefile:179: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/include/toplevel.mk:248: world] Error 2

VIKINGYFY, 2024-11-24:
自己排查加的插件

qqsir-dev, 2024-11-24:
我是这么加的，应该是不对，但是不知道怎么改了，麻烦给看看：
UPDATE_PACKAGE "OpenAppFilter" "destan19/OpenAppFilter" "master" "pkg"
UPDATE_PACKAGE "ttyd" "coolsnowwolf/packages/utils/ttyd" "master" "pkg"

然后就看步骤到日志里有这个问题
cp: missing destination file operand after './'
Try 'cp --help' for more information.
Cloning into 'ttyd'...
remote: Not Found
fatal: repository 'https://github.com/coolsnowwolf/packages/utils/ttyd.git/' not found
find: ‘./packages/*/’: No such file or directory
cp: missing destination file operand after './'
Try 'cp --help' for more information.

VIKINGYFY, 2024-11-24:
UPDATE_PACKAGE "ttyd" "coolsnowwolf/packages/utils/ttyd" "master" "pkg"

写错了，认真看上面的注释

qqsir-dev, 2024-11-24:
认真看了，但是不太了解原理，改了一下好像还是不行
UPDATE_PACKAGE "ttyd" "coolsnowwolf/packages/utils" "master"
日志：
Cloning into 'luci-app-tailscale'...
Cloning into 'packages'...
Cloning into 'OpenAppFilter'...
cp: missing destination file operand after './'
Try 'cp --help' for more information.
fatal: destination path 'utils' already exists and is not an empty directory.

qqsir-dev, 2024-11-24:
又改了下：
UPDATE_PACKAGE "utils/ttyd" "coolsnowwolf/packages" "master"
日志：
Cloning into 'packages'...
Cloning into 'OpenAppFilter'...
cp: missing destination file operand after './'
Try 'cp --help' for more information.
Cloning into 'packages'...
 
sing-box version update has started!
1.10.2 v1.10.2 1.10.2 cae964985e6f27a1ddefc074849a950[43](https://github.com/qqsir-dev/OpenWRT-CI/actions/runs/11995530419/job/33438904733#step:11:44)7e27ef5bfb44847d5ce5d5b2a1b7a27
../feeds/packages/net/sing-box/Makefile version is already the latest!

qqsir-dev, 2024-11-24:
哦抱歉，好像不用克隆ttyd，不过即便是不克隆ttyd，日志仍有cp的报警
Cloning into 'OpenAppFilter'...
cp: missing destination file operand after './'
Try 'cp --help' for more information.
 
sing-box version update has started!
1.10.2 v1.10.2 1.10.2 cae964985e6f27a1ddefc074849a950[43](https://github.com/qqsir-dev/OpenWRT-CI/actions/runs/11995686821/job/33439240271#step:11:44)7e27ef5bfb[44](https://github.com/qqsir-dev/OpenWRT-CI/actions/runs/11995686821/job/33439240271#step:11:45)847d5ce5d5b2a1b7a27
../feeds/packages/net/sing-box/Makefile version is already the latest!

VIKINGYFY, 2024-11-24:
上面那个不要用pkg参数，

qqsir-dev, 2024-11-24:
额。。去掉后 克隆日志没错了，自定义设置里日志又又问题：之前一出现这个后面就会报上面coremark的错误
tmp/.config-package.in:6661:error: recursive dependency detected!
tmp/.config-package.in:6661:	symbol PACKAGE_kmod-oaf is selected by PACKAGE_kmod-oaf
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

VIKINGYFY, 2024-11-24:
你加的OpenAppFilter插件依赖有报错

qqsir-dev, 2024-11-24:
是不是本项目的包如果也是从immortalwrt/packages来的话，我也不需要克隆其他源的包，直接在配置文件里加上oaf相应的设置就行了

qqsir-dev, 2024-11-24:
多谢耐心指点，你这个项目写的很干练，语言基本能看明白，就是水平有限，得一点一点琢磨。

closed 2024-11-25 by VIKINGYFY

master-yun-yun, 2024-12-01:
因此，最终正确的是怎么写的？