VIKINGYFY/OpenWRT-CI #105: 如何编译alist

master-yun-yun, 2024-12-01:
大佬
你代码已经有alist源，但写的是 alist 不是 luci-app-alist
编译进去是增加
1）CONFIG_PACKAGE_luci-app-alist=y
还是
2）CONFIG_PACKAGE_alist=y
还是说下面2行要一起加？
1）CONFIG_PACKAGE_luci-app-alist=y
2）CONFIG_PACKAGE_alist=y

master-yun-yun, 2024-12-01:
另外
ipk插件格式前，自定义源编译成功
改为apk后，同样的写法格式，编译总是失败
一堆error，帮看看是哪里问题
for mod in /home/runner/work/JDC-AX1800-Pro/JDC-AX1800-Pro/wrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.6.63/crypto/crypto_null.ko; do if grep -q "${mod##/home/runner/work/JDC-AX1800-Pro/JDC-AX1800-Pro/wrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.6.63/}" "/home/runner/work/JDC-AX1800-Pro/JDC-AX1800-Pro/wrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.6.63/modules.builtin"; then echo "NOTICE: module '$mod' is built-in."; elif [ -e $mod ]; then mkdir -p /home/runner/work/JDC-AX1800-Pro/JDC-AX1800-Pro/wrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-null/lib/modules/6.6.63 ; cp -fpR -L $mod /home/runner/work/JDC-AX1800-Pro/JDC-AX1800-Pro/wrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-null/lib/modules/6.6.63/ ; _**else echo "ERROR: module '$mod' is missing." >&2; exit 1; fi; done;**_

VIKINGYFY, 2024-12-01:
CONFIG_PACKAGE_luci-app-alist=y
写这个就行，核心会自动勾选

closed 2024-12-03 by VIKINGYFY

master-yun-yun, 2024-12-08:
> CONFIG_PACKAGE_luci-app-alist=y 写这个就行，核心会自动勾选
3种组合都试过了，编译失败，提示：
**Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on**
另外，只要选了luci-app-openvpn-server,必定失败
能帮忙看看吗？
谢谢