VIKINGYFY/OpenWRT-CI #106: ZyXEL EX5700 (Telenor),Wifi in 6GHz mode does not work

Doanduy09, 2024-12-05:
ZyXEL EX5700 (Telenor), you can add the option of 6ghz wifi with additional selection of 5ghz or 6ghz band... ZyXEL EX5700 (Telenor). When I turn it on in the mode as shown in the picture, the laptop and phone cannot find it. Wifi in 6GHz mode does not work. Can anyone tell me how to change it to 5GHz mode for it to work? Thank you.
[image]

VIKINGYFY, 2024-12-05:
WIFI country chang to the US，try it

Doanduy09, 2024-12-05:
> Nước WIFI đổi sang Mỹ, thử xem

i switched to us region like it doesnt work

VIKINGYFY, 2024-12-05:
please use CloseWRT firmware, 6G support is very good.
https://github.com/VIKINGYFY/CloseWRT-CI/tags

closed 2024-12-09 by VIKINGYFY