VIKINGYFY/OpenWRT-CI #118: HomeProxy (sing-box v1.10.6) 未运行

yeqifeng2288, 2025-01-08:
[IPQ60XX-WIFI-YES_immortalwrt.git-main_25.01.08_20.22.55](https://github.com/VIKINGYFY/OpenWRT-CI/releases/tag/IPQ60XX-WIFI-YES_immortalwrt.git-main_25.01.08_20.22.55)

HomeProxy一直提示这个，有人遇到同样的问题吗？
[immortalwrt.git-main_jdcloud_re-ss-01-squashfs-sysupgrade_25.01.08_20.22.55.bin](https://github.com/VIKINGYFY/OpenWRT-CI/releases/download/IPQ60XX-WIFI-YES_immortalwrt.git-main_25.01.08_20.22.55/immortalwrt.git-main_jdcloud_re-ss-01-squashfs-sysupgrade_25.01.08_20.22.55.bin)

www78814, 2025-01-08:
我也是刚刷的[IPQ60XX-WIFI-YES_immortalwrt.git-main_25.01.08_20.22.55](https://github.com/VIKINGYFY/OpenWRT-CI/releases/download/IPQ60XX-WIFI-YES_immortalwrt.git-main_25.01.08_20.22.55/immortalwrt.git-main_zn_m2-squashfs-factory_25.01.08_20.22.55.ubi)固件，homeproxy也无法启动，luci界面日志无显示。
在ssh里手动启动提示如下：
root@OWRT:~# /etc/init.d/homeproxy start
-ash: /etc/init.d/homeproxy: Permission denied

VIKINGYFY, 2025-01-09:
好了，权限丢了

closed 2025-01-09 by VIKINGYFY