VIKINGYFY/OpenWRT-CI #138: 亚瑟ax1800pro 无线中继，路由器无法作为客户端（sta）连上WiFi

luyud, 2025-03-01:
试了很多WiFi， ac，ax, bgn模式均不能正常连上或获取到IP地址（上游WiFi有dhcp） 日志一直重复看着像不停的在断联， 请问下是否知道什么原因或者是缺少什么驱动导致的，我试了很多类似的固件都不行，但是这些固件都几乎来自官方的源， 

QWRT是可以连上的，但是内核太旧了很多插件用不了

```
Sat Mar  1 22:15:41 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: CTRL-EVENT-DISCONNECTED bssid=7c:b5:9b:3b:83:c2 reason=3
Sat Mar  1 22:15:41 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Added BSSID 7c:b5:9b:3b:83:c2 into ignore list, ignoring for 10 seconds
Sat Mar  1 22:15:42 2025 daemon.warn odhcpd[2379]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Sat Mar  1 22:15:42 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan1)
Sat Mar  1 22:15:42 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan2)
Sat Mar  1 22:15:42 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan3)
Sat Mar  1 22:15:42 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy0-sta0)
Sat Mar  1 22:15:42 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy1-ap0)
Sat Mar  1 22:15:42 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (wan)
Sat Mar  1 22:15:42 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Removed BSSID 7c:b5:9b:3b:83:c2 from ignore list (clear)
Sat Mar  1 22:15:42 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: SME: Trying to authenticate with 7c:b5:9b:3b:83:c2 (SSID='TP-LINK_5G_83C0' freq=5805 MHz)
Sat Mar  1 22:15:42 2025 kern.info kernel: [ 1215.906939] phy0-sta0: authenticate with 7c:b5:9b:3b:83:c2 (local address=de:d8:7c:49:4f:9e)
Sat Mar  1 22:15:42 2025 kern.info kernel: [ 1215.906995] phy0-sta0: send auth to 7c:b5:9b:3b:83:c2 (try 1/3)
Sat Mar  1 22:15:42 2025 kern.info kernel: [ 1215.916294] phy0-sta0: authenticated
Sat Mar  1 22:15:42 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Trying to associate with 7c:b5:9b:3b:83:c2 (SSID='TP-LINK_5G_83C0' freq=5805 MHz)
Sat Mar  1 22:15:42 2025 kern.info kernel: [ 1215.926783] phy0-sta0: associate with 7c:b5:9b:3b:83:c2 (try 1/3)
Sat Mar  1 22:15:42 2025 daemon.notice netifd: Network device 'phy0-sta0' link is up
Sat Mar  1 22:15:42 2025 daemon.notice netifd: Interface 'wwan' has link connectivity
Sat Mar  1 22:15:42 2025 daemon.notice netifd: Interface 'wwan' is setting up now
Sat Mar  1 22:15:42 2025 kern.info kernel: [ 1215.938951] phy0-sta0: RX AssocResp from 7c:b5:9b:3b:83:c2 (capab=0x11 status=0 aid=1)
Sat Mar  1 22:15:42 2025 kern.info kernel: [ 1215.940371] phy0-sta0: associated
Sat Mar  1 22:15:42 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Associated with 7c:b5:9b:3b:83:c2
Sat Mar  1 22:15:42 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sat Mar  1 22:15:42 2025 daemon.notice netifd: wwan (9539): udhcpc: started, v1.37.0
Sat Mar  1 22:15:42 2025 daemon.notice netifd: wwan (9539): udhcpc: broadcasting discover
Sat Mar  1 22:15:43 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: WPA: Key negotiation completed with 7c:b5:9b:3b:83:c2 [PTK=CCMP GTK=CCMP]
Sat Mar  1 22:15:43 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: CTRL-EVENT-CONNECTED - Connection to 7c:b5:9b:3b:83:c2 completed [id=0 id_str=]
Sat Mar  1 22:15:46 2025 daemon.notice netifd: wwan (9539): udhcpc: broadcasting discover
Sat Mar  1 22:15:49 2025 daemon.notice netifd: wwan (9539): udhcpc: broadcasting discover
Sat Mar  1 22:23:39 2025 daemon.warn odhcpd[2379]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Sat Mar  1 22:24:00 2025 daemon.notice netifd: Network device 'phy0-sta0' link is down
Sat Mar  1 22:24:00 2025 daemon.notice netifd: Interface 'wwan' has link connectivity loss
Sat Mar  1 22:24:00 2025 kern.info kernel: [ 1713.357003] phy0-sta0: deauthenticated from 7c:b5:9b:3b:83:c2 (Reason: 3=DEAUTH_LEAVING)
Sat Mar  1 22:24:00 2025 daemon.notice netifd: wwan (9539): udhcpc: received SIGTERM
Sat Mar  1 22:24:00 2025 daemon.notice netifd: wwan (9539): udhcpc: entering released state
Sat Mar  1 22:24:00 2025 daemon.notice netifd: wwan (9539): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wwan" } (Permission denied)
Sat Mar  1 22:24:00 2025 daemon.notice netifd: Interface 'wwan' is now down
Sat Mar  1 22:24:00 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: CTRL-EVENT-DISCONNECTED bssid=7c:b5:9b:3b:83:c2 reason=3
Sat Mar  1 22:24:00 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Added BSSID 7c:b5:9b:3b:83:c2 into ignore list, ignoring for 10 seconds
Sat Mar  1 22:24:01 2025 daemon.warn odhcpd[2379]: A default route is present but there is no public prefix on lan thus we announce no default route by setting ra_lifetime to 0!
Sat Mar  1 22:24:01 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan1)
Sat Mar  1 22:24:01 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan2)
Sat Mar  1 22:24:01 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan3)
Sat Mar  1 22:24:01 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy0-sta0)
Sat Mar  1 22:24:01 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy1-ap0)
Sat Mar  1 22:24:01 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (wan)
Sat Mar  1 22:24:01 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Removed BSSID 7c:b5:9b:3b:83:c2 from ignore list (clear)
Sat Mar  1 22:24:01 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: SME: Trying to authenticate with 7c:b5:9b:3b:83:c2 (SSID='TP-LINK_5G_83C0' freq=5805 MHz)
Sat Mar  1 22:24:01 2025 kern.info kernel: [ 1714.897347] phy0-sta0: authenticate with 7c:b5:9b:3b:83:c2 (local address=de:d8:7c:49:4f:9e)
Sat Mar  1 22:24:01 2025 kern.info kernel: [ 1714.897402] phy0-sta0: send auth to 7c:b5:9b:3b:83:c2 (try 1/3)
Sat Mar  1 22:24:01 2025 kern.info kernel: [ 1714.906464] phy0-sta0: authenticated
Sat Mar  1 22:24:01 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Trying to associate with 7c:b5:9b:3b:83:c2 (SSID='TP-LINK_5G_83C0' freq=5805 MHz)
Sat Mar  1 22:24:01 2025 kern.info kernel: [ 1714.916864] phy0-sta0: associate with 7c:b5:9b:3b:83:c2 (try 1/3)
Sat Mar  1 22:24:01 2025 daemon.notice netifd: Network device 'phy0-sta0' link is up
Sat Mar  1 22:24:01 2025 daemon.notice netifd: Interface 'wwan' has link connectivity
Sat Mar  1 22:24:01 2025 daemon.notice netifd: Interface 'wwan' is setting up now
Sat Mar  1 22:24:01 2025 kern.info kernel: [ 1714.929219] phy0-sta0: RX AssocResp from 7c:b5:9b:3b:83:c2 (capab=0x11 status=0 aid=1)
Sat Mar  1 22:24:01 2025 kern.info kernel: [ 1714.930426] phy0-sta0: associated
Sat Mar  1 22:24:01 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: Associated with 7c:b5:9b:3b:83:c2
Sat Mar  1 22:24:01 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sat Mar  1 22:24:01 2025 daemon.notice netifd: wwan (9870): udhcpc: started, v1.37.0
Sat Mar  1 22:24:01 2025 daemon.notice netifd: wwan (9870): udhcpc: broadcasting discover
Sat Mar  1 22:24:02 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: WPA: Key negotiation completed with 7c:b5:9b:3b:83:c2 [PTK=CCMP GTK=CCMP]
Sat Mar  1 22:24:02 2025 daemon.notice wpa_supplicant[3672]: phy0-sta0: CTRL-EVENT-CONNECTED - Connection to 7c:b5:9b:3b:83:c2 completed [id=0 id_str=]
Sat Mar  1 22:24:05 2025 daemon.notice netifd: wwan (9870): udhcpc: broadcasting discover
Sat Mar  1 22:24:08 2025 daemon.notice netifd: wwan (9870): udhcpc: broadcasting discover
```

VIKINGYFY, 2025-03-01:
ath11k + nss的问题，目前无解

luyud, 2025-03-01:
啊，好吧，这就太难受了

simonchan6688, 2025-03-21:
同样的问题搜索来到这里。实测回退到2025年1月之前的固件可以正常使用无限客户端模式

rebron1900, 2025-03-24:
> 同样的问题搜索来到这里。实测回退到2025年1月之前的固件可以正常使用无线客户端模式

确实，我这边刷了一个12月份左右的包，能正常使用。

经过测试，owrt、libwrt、kwrt的新版本都没办法正常中继，症状为能连上wifi，但是分配不到ip

luyud, 2025-03-27:
我发现这好像是个和 openwrt 自己的 dhcp 兼容性问题,  我把同一个固件连接家里的 Wi-Fi(openwrt 主路由ImmortalWrt 24.10.0)就无法分配到 ip, 但是我连接公司的 Wi-Fi 就没问题了.  但我没有尝试更多的路由器

VIKINGYFY, 2025-03-31:
已根据q佬回复删除了wds补丁，明天测试一下看看

VIKINGYFY, 2025-04-01:
经群友测试，已修复，中继可正常获取到v4 v6。

closed 2025-04-01 by VIKINGYFY

GHNERCH, 2025-04-12:
> 已根据q佬回复删除了wds补丁，明天测试一下看看

您好，v大，能不能详细说明一下如何删除wds补丁，解决亚瑟ax1800pro无线中继的问题，您的固件很好用，可我也想尝试6.12内核的固件，听说优化了内存管理，静候佳音！

GHNERCH, 2025-04-12:
> 已根据q佬回复删除了wds补丁，明天测试一下看看

还有个小问题，kiddin9大佬的固件，CPU貌似可以超频到1.8ghz,不知是真是假，希望可以加入相关代码，谢谢您！

VIKINGYFY, 2025-04-12:
> > 已根据q佬回复删除了wds补丁，明天测试一下看看
> 
> 还有个小问题，kiddin9大佬的固件，CPU貌似可以超频到1.8ghz,不知是真是假，希望可以加入相关代码，谢谢您！

ipq6000最高频率就1.5g，6010才是1.8g，你自己可以试试跑分就懂了