VIKINGYFY/OpenWRT-CI #152: dsa存在fdb学习错误问题

hjchjchjc4352, 2025-04-05:
我使用多个雅典娜ax6600组网，划分了vlan开启802.11kvr漫游，但存在不同ap漫游时候无法连接ap的问题，只能等待5分钟后bridge fdb条目自动刷新才能恢复。类似这几个链接下的bug
https://github.com/openwrt/openwrt/issues/11650
https://github.com/rany2/openwrt/commit/b05a7f79787cbc9020f80f8e315c09f37ffdec9c
目前是通过修改vlan设备的mac老化时间改为3秒2秒暂时可以漫游，但存在短时间丢包断连问题，希望大佬有时候可以看下。。。

hjchjchjc4352, 2025-04-10:
通过https://github.com/openwrt/openwrt/issues/11650下的一个脚本临时解决了，现在漫游也没问题了。
还是想请问下大佬，固件安装了igmpproxy运行提示MRT_DEL_MFC; Errno(2): No such file or directory，iptv组播也无法播放，请问大佬是内核多播路由的问题么。。。

VIKINGYFY, 2025-04-16:
组播这个不清楚，大概率是跟内核有关系，之前有人也提过组播问题

另外，那个脚本内容发一下，我看看

hjchjchjc4352, 2025-04-17:
> 组播这个不清楚，大概率是跟内核有关系，之前有人也提过组播问题
> 
> 另外，那个脚本内容发一下，我看看

#!/bin/sh
# 本脚本用于监控 OpenWrt 中 FDB 条目的变化，
# 针对嵌套桥接环境进行判断，避免对合法的嵌套记录执行删除操作。
#
# 示例合法情况：
#
# VLAN 88：
#   ① e.g. 68:df:82:9f:6f:6a dev wan vlan 88 master br-mine
#       与 e.g. 68:df:82:9f:6f:6a dev br-mine.88 master br-lan
#   ② e.g. a0:cc:f8:76:a0:92 dev lan3 vlan 88 master br-mine
#       与 e.g. a0:cc:f8:76:a0:92 dev br-mine.88 master br-lan
#
# VLAN 90：
#   ③ e.g. 68:df:82:9f:6f:6a dev wan vlan 90 master br-mine
#       与 e.g. 68:df:82:9f:6f:6a dev br-mine.90 master br-guest
#   ④ e.g. a0:cc:f8:76:a0:92 dev lan3 vlan 90 master br-mine
#       与 e.g. a0:cc:f8:76:a0:92 dev br-mine.90 master br-guest
#
# 对于不符合上述嵌套关系的 FDB 条目，脚本会尝试删除旧记录。

# 定义需要监控的桥接设备
bridge_devices="br-mine wan lan1 lan2 lan3 lan4"

# 监听 FDB 事件（需系统支持 bridge monitor fdb 命令）
/usr/sbin/bridge monitor fdb | while read -r mac d dev rest; do
  if [ "$mac" != "Deleted" ]; then
    echo "在端口 $dev 上发现新 MAC: $mac"
    for bridge_device in $bridge_devices; do
      /usr/sbin/bridge fdb show br "$bridge_device" | while read -r omac dummy odev rest2; do
        # 当同一 MAC 在不同接口出现时
        if [ "$omac" = "$mac" ] && [ "$odev" != "$dev" ]; then

          ###### VLAN 88 合法嵌套判断 ######
          # 条件1: 在 wan 接口上，且记录包含 "vlan 88" 和 "master br-mine"，新记录在 br-mine.88
          if [ "$odev" = "wan" ] && \
             echo "$rest2" | grep -q "vlan 88" && \
             echo "$rest2" | grep -q "master br-mine" && \
             [ "$dev" = "br-mine.88" ]; then
            echo "合法嵌套 VLAN 88: $odev -> $dev (wan -> br-mine.88)"
            continue
          fi

          # 条件2: 在物理 LAN 接口（lanX）上，记录包含 "vlan 88" 和 "master br-mine"，新记录在 br-mine.88
          if echo "$odev" | grep -E '^lan[0-9]+' >/dev/null 2>&1 && \
             echo "$rest2" | grep -q "vlan 88" && \
             echo "$rest2" | grep -q "master br-mine" && \
             [ "$dev" = "br-mine.88" ]; then
            echo "合法嵌套 VLAN 88: $odev -> $dev (LAN -> br-mine.88)"
            continue
          fi

          # 条件3: 在 br-mine.88 上（记录包含 "master br-lan"），新记录出现在物理 LAN 接口（lanX）
          if [ "$odev" = "br-mine.88" ] && \
             echo "$rest2" | grep -q "master br-lan" && \
             echo "$dev" | grep -E '^lan[0-9]+' >/dev/null 2>&1; then
            echo "合法嵌套 VLAN 88: $odev -> $dev (br-mine.88 -> LAN)"
            continue
          fi

          # 条件4: 在 br-mine.88 上（记录包含 "master br-lan"），新记录在 br-lan
          if [ "$odev" = "br-mine.88" ] && \
             echo "$rest2" | grep -q "master br-lan" && \
             [ "$dev" = "br-lan" ]; then
            echo "合法嵌套 VLAN 88: $odev -> $dev (br-mine.88 -> br-lan)"
            continue
          fi

          ###### VLAN 90 合法嵌套判断 ######
          # 条件5: 在 wan 接口上，记录包含 "vlan 90" 和 "master br-mine"，新记录在 br-mine.90
          if [ "$odev" = "wan" ] && \
             echo "$rest2" | grep -q "vlan 90" && \
             echo "$rest2" | grep -q "master br-mine" && \
             [ "$dev" = "br-mine.90" ]; then
            echo "合法嵌套 VLAN 90: $odev -> $dev (wan -> br-mine.90)"
            continue
          fi

          # 条件6: 在 br-mine.90 上（记录包含 "master br-guest"），新记录在 br-guest
          if [ "$odev" = "br-mine.90" ] && \
             echo "$rest2" | grep -q "master br-guest" && \
             [ "$dev" = "br-guest" ]; then
            echo "合法嵌套 VLAN 90: $odev -> $dev (br-mine.90 -> br-guest)"
            continue
          fi

          # 条件7: 在物理 LAN 接口（lanX）上，记录包含 "vlan 90" 和 "master br-mine"，新记录在 br-mine.90
          if echo "$odev" | grep -E '^lan[0-9]+' >/dev/null 2>&1 && \
             echo "$rest2" | grep -q "vlan 90" && \
             echo "$rest2" | grep -q "master br-mine" && \
             [ "$dev" = "br-mine.90" ]; then
            echo "合法嵌套 VLAN 90: $odev -> $dev (LAN -> br-mine.90)"
            continue
          fi

          # 条件8: 在 br-mine.90 上（记录包含 "master br-guest"），新记录出现在物理 LAN 接口（lanX）
          if [ "$odev" = "br-mine.90" ] && \
             echo "$rest2" | grep -q "master br-guest" && \
             echo "$dev" | grep -E '^lan[0-9]+' >/dev/null 2>&1; then
            echo "合法嵌套 VLAN 90: $odev -> $dev (br-mine.90 -> LAN)"
            continue
          fi

          ##### 非合法嵌套，执行删除操作 #####
          # 为确保删除命令格式正确，将 "master <设备>" 替换为仅保留 "master"
          clean_rest=$(echo "$rest2" | sed -e 's/\(master\) [^ ]*/\1/g')
          echo "在桥接接口 $bridge_device 上发现 MAC $mac 在接口 $odev 的记录，但应在接口 $dev，此时尝试删除旧记录"
          output=$(/usr/sbin/bridge fdb del "$mac" dev "$odev" $clean_rest 2>&1)
          if echo "$output" | grep -q "No such file or directory"; then
            echo "MAC $mac 在接口 $odev 的旧记录已不存在"
          elif [ -n "$output" ]; then
            echo "删除 MAC $mac 在 $odev 时出错: $output"
          fi

        fi
      done
    done
  fi
done

脚本是用bingai写的，我试了有用就这么来自启动了

hjchjchjc4352, 2025-04-17:
> 组播这个不清楚，大概率是跟内核有关系，之前有人也提过组播问题
> 
> 另外，那个脚本内容发一下，我看看

config device
	option type 'bridge'
	option name 'br-mine'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config bridge-vlan
	option device 'br-mine'
	option vlan '88'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config bridge-vlan
	option device 'br-mine'
	option vlan '90'
	list ports 'lan4:t'
	list ports 'wan:t'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'br-mine.88'

这是我的网络配置

closed 2025-05-20 by hjchjchjc4352

naoki66, 2025-09-01:
mark