VIKINGYFY/OpenWRT-CI #153: 雲編譯CR1000A 網口無法正常使用

z21100644, 2025-04-11:
wan口自動協商異常,使用dhcp或pppoe都無法獲取上游IP
[   62.568036] nss-dp 3a007000.dp6-syn wan: PHY Link up speed: 1000
[   63.388773] uniphy autoneg time out!

lan口也許是驅動的問題,無法連接上

內核日誌:
<details><summary>Details</summary>
<p>

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.85 (runner@fv-az1778-511) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0-62cab9f) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP PREEMPT Thu Apr 10 14:39:23 2025
[    0.000000] Machine model: Verizon CR1000A
[    0.000000] OF: reserved mem: 0x0000000040000000..0x0000000040ffffff (16384 KiB) nomap non-reusable nss@40000000
[    0.000000] OF: reserved mem: 0x000000004a400000..0x000000004a5fffff (2048 KiB) nomap non-reusable tzapp@4a400000
[    0.000000] OF: reserved mem: 0x000000004a600000..0x000000004a9fffff (4096 KiB) nomap non-reusable bootloader@4a600000
[    0.000000] OF: reserved mem: 0x000000004aa00000..0x000000004aafffff (1024 KiB) nomap non-reusable sbl@4aa00000
[    0.000000] OF: reserved mem: 0x000000004ab00000..0x000000004abfffff (1024 KiB) nomap non-reusable smem@4ab00000
[    0.000000] OF: reserved mem: 0x000000004ac00000..0x000000004affffff (4096 KiB) nomap non-reusable memory@4ac00000
[    0.000000] OF: reserved mem: 0x000000004b000000..0x000000004effffff (65536 KiB) nomap non-reusable q6@4b000000
[    0.000000] OF: reserved mem: 0x000000004f000000..0x000000004f0fffff (1024 KiB) nomap non-reusable q6_etr@4f000000
[    0.000000] OF: reserved mem: 0x000000004f100000..0x000000004f1fffff (1024 KiB) nomap non-reusable m3_dump@4f100000
[    0.000000] OF: reserved mem: 0x000000004f200000..0x000000004f2fffff (1024 KiB) nomap non-reusable ramoops@4f200000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x000000004f2fffff]
[    0.000000]   node   0: [mem 0x000000004f300000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 18 pages/cpu s36584 r8192 d28952 u73728
[    0.000000] pcpu-alloc: s36584 r8192 d28952 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 root=/dev/mmcblk0p20 rootwait fstools_ignore_partname=1
[    0.000000] Unknown kernel command line parameters "fstools_ignore_partname=1", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516096
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 2MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000bd600000-0x00000000bd800000] (2MB)
[    0.000000] Memory: 1943548K/2097152K available (8448K kernel code, 920K rwdata, 2556K rodata, 1536K init, 280K bss, 153604K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000106] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000119] pid_max: default: 32768 minimum: 301
[    0.005388] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.005406] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[... 496 lines cut ...]
[   72.716281] br-lan: port 3(phy2-ap0) entered forwarding state
[   72.724479] ath11k_pci 0000:01:00.0 phy2-ap0: left allmulticast mode
[   72.727693] ath11k_pci 0000:01:00.0 phy2-ap0: left promiscuous mode
[   72.734263] br-lan: port 3(phy2-ap0) entered disabled state
[   72.830707] br-lan: port 3(phy2-ap0) entered blocking state
[   72.830772] br-lan: port 3(phy2-ap0) entered disabled state
[   72.835173] ath11k_pci 0000:01:00.0 phy2-ap0: entered allmulticast mode
[   72.841328] ath11k_pci 0000:01:00.0 phy2-ap0: entered promiscuous mode
[   72.847690] br-lan: port 3(phy2-ap0) entered blocking state
[   72.853859] br-lan: port 3(phy2-ap0) entered forwarding state
[   73.010010] br-lan: port 4(phy0-ap0) entered blocking state
[   73.010052] br-lan: port 4(phy0-ap0) entered disabled state
[   73.014485] ath11k c000000.wifi phy0-ap0: entered allmulticast mode
[   73.020475] ath11k c000000.wifi phy0-ap0: entered promiscuous mode
[   73.026339] br-lan: port 4(phy0-ap0) entered blocking state
[   73.032487] br-lan: port 4(phy0-ap0) entered forwarding state
[   73.038423] br-lan: port 3(phy2-ap0) entered disabled state
[   73.044130] br-lan: port 4(phy0-ap0) entered disabled state
[  136.240839] br-lan: port 4(phy0-ap0) entered blocking state
[  136.240886] br-lan: port 4(phy0-ap0) entered forwarding state
[  136.372604] br-lan: port 3(phy2-ap0) entered blocking state
[  136.372656] br-lan: port 3(phy2-ap0) entered forwarding state

</p>
</details>

VIKINGYFY, 2025-04-11:
自己编译的话，nss-macsec不要选，会导致高速率网口协商失败

z21100644, 2025-04-11:
確認了一下 Config裡沒選nss-macsec

[Config_IPQ807X-WIFI-YES_immortalwrt.git-main_25.04.11_11.32.32.txt](https://github.com/user-attachments/files/19701231/Config_IPQ807X-WIFI-YES_immortalwrt.git-main_25.04.11_11.32.32.txt)

VIKINGYFY, 2025-04-11:
那就不清楚了，这个机器是群友提的pr适配的，我没有机器测试

kinglove1990, 2025-04-11:
需要用脚本，启动RTL usrapp 才能正确识别lan口

kinglove1990, 2025-04-11:
V佬的固件应该能自动识别wan口啊  只不过lan口不能识别啊

z21100644, 2025-04-11:
我也試過大佬你的測試固件，然而情況還是相同 wan口異常

kinglove1990, 2025-04-11:
> 我也試過大佬你的測試固件，然而情況還是相同 wan口異常

昨天试了一下，无线中继和lan口正常，wan口确实异常，试了一下遗留的6.6.6x版本都还是正常的！去查查原因！

z21100644, 2025-04-12:
wan口能連接上
```
root@OWRT:~# dmesg | grep -i aquantia
[    4.432593] Aquantia AQR113C 90000.mdio-1:08: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:08, irq=POLL)

[   58.897755] nss-dp 3a007000.dp6-syn wan: PHY Link up speed: 1000
[   59.308449] uniphy autoneg time out!
``` 
但無法獲得IP的問題 我是真沒頭緒
```
root@OWRT:~# cat /sys/bus/mdio_bus/devices/90000.mdio-1:08/phy_id
0x00000000

[   58.897755] nss-dp 3a007000.dp6-syn wan: PHY Link up speed: 1000
[   59.308449] uniphy autoneg time out!
```

kinglove1990, 2025-04-23:
> wan口能連接上
> 
> ```
> root@OWRT:~# dmesg | grep -i aquantia
> [    4.432593] Aquantia AQR113C 90000.mdio-1:08: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:08, irq=POLL)
> ```
> 
> 但無法獲得IP的問題 我是真沒頭緒
> 
> ```
> root@OWRT:~# cat /sys/bus/mdio_bus/devices/90000.mdio-1:08/phy_id
> 0x00000000
> 
> [   58.897755] nss-dp 3a007000.dp6-syn wan: PHY Link up speed: 1000
> [   59.308449] uniphy autoneg time out!
> ```

DTS修正一下就好了 

[image]

VIKINGYFY, 2025-04-24:
已合并修复

closed 2025-04-24 by VIKINGYFY