VIKINGYFY/OpenWRT-CI #158: WARNING: Makefile 'package

iFHax, 2025-04-22:
Collecting target info: doneWARNING: Makefile 'package/utils/audit/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/emortal/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/emortal/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/emortal/default-settings/Makefile' has a dependency on 'luci-i18n-base-zh-cn', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/emortal/luci-app-athena-led/Makefile' has a dependency on 'luci-lua-runtime', which does not exist
WARNING: Makefile 'package/emortal/luci-app-athena-led/Makefile' has a dependency on 'luci-lua-runtime', which does not exist
WARNING: Makefile 'package/emortal/luci-app-athena-led/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/emortal/luci-app-athena-led/Makefile' has a build dependency on 'csstidy/host', which does not exist
WARNING: Makefile 'package/emortal/luci-app-athena-led/Makefile' has a build dependency on 'luasrcdiet/host', which does not exist
WARNING: Makefile 'package/utils/pcat-manager/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/utils/pcat-manager/Makefile' has a dependency on 'libgpiod', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist

VIKINGYFY, 2025-04-23:
warning不是error，一般情况无视就行，另外，你这是本地编译吧？包都没下载全

VIKINGYFY, 2025-04-23:
我看了你fork后的项目，并没有出现类似提示，使用别人的项目请自行解决

closed 2025-04-23 by VIKINGYFY