VIKINGYFY/OpenWRT-CI #159: 你好，请问这个支持 Arcadyan AW1000 路由器吗？另外，支持 USB 上网卡功能吗？

intannajwa, 2025-04-23:
你好，请问这个支持 Arcadyan AW1000 路由器吗？另外，支持 USB 上网卡功能吗？

VIKINGYFY, 2025-04-23:
设备支持，usb上网已添加常见模块，但没有实机测试，自己编译的话可以加上qmode

intannajwa, 2025-04-23:
**你好，我在 `./wrt/` 目录下运行 `make` 时，出现了这个警告：**

```
WARNING: Makefile 'package/modem_feeds/luci/luci-app-qmodem/Makefile' has a dependency on 'quectel-CM-5G', which does not exist
```

**编译还是继续进行，比如 `make[1] world`、`make[2] tools/compile` 都没有报错。我想确认一下，这个警告可以忽略吗？还是说缺少某个软件包？谢谢！**

intannajwa, 2025-04-23:
我刚做了一些测试，结果如下：
IPQ807X-WIFI-YES-AW1000 – 使用原版 GENERAL.txt 配置 – 正常工作
IPQ807X-WIFI-NO-AW1000 – 使用原版 GENERAL.txt 配置 – 无法启动（变砖）

VIKINGYFY, 2025-04-23:
> **你好，我在 `./wrt/` 目录下运行 `make` 时，出现了这个警告：**
> 
> ```
> WARNING: Makefile 'package/modem_feeds/luci/luci-app-qmodem/Makefile' has a dependency on 'quectel-CM-5G', which does not exist
> ```
> 
> **编译还是继续进行，比如 `make[1] world`、`make[2] tools/compile` 都没有报错。我想确认一下，这个警告可以忽略吗？还是说缺少某个软件包？谢谢！**

可以无视，我说的就是添加上这个luci-app-qmodem

VIKINGYFY, 2025-04-24:
> make[3] -C feeds/packages/lang/rust host-compile ERROR: package/feeds/packages/gn [host] failed to build. make[3] -C package/system/apk host-compile

你选了啥插件？gn版本报错了

intannajwa, 2025-04-24:
我刚刚添加了这个

Passwall
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_v2ray-core=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_trojan-go=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_hysteria=y
CONFIG_PACKAGE_chinadns-ng=y
CONFIG_PACKAGE_dns2socks=y
CONFIG_PACKAGE_ipt2socks=y
CONFIG_PACKAGE_microsocks=y
CONFIG_PACKAGE_tuic-client=y
CONFIG_PACKAGE_naiveproxy=y
CONFIG_PACKAGE_https-dns-proxy=y

CONFIG_PACKAGE_luci-app-passwall2=y
CONFIG_PACKAGE_luci-app-openclash=y

# Compress image
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_PACKAGE_luci-app-irqbalance=y
CONFIG_PACKAGE_irqbalance=y # https://github.com/4IceG/luci-app-irqbalance

intannajwa, 2025-04-24:
> > make[3] -C feeds/packages/lang/rust host-compile ERROR: package/feeds/packages/gn [host] failed to build. make[3] -C package/system/apk host-compile
> 
> 你选了啥插件？gn版本报错了

我刚刚添加了这个

Passwall
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_v2ray-core=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_trojan-go=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_hysteria=y
CONFIG_PACKAGE_chinadns-ng=y
CONFIG_PACKAGE_dns2socks=y
CONFIG_PACKAGE_ipt2socks=y
CONFIG_PACKAGE_microsocks=y
CONFIG_PACKAGE_tuic-client=y
CONFIG_PACKAGE_naiveproxy=y
CONFIG_PACKAGE_https-dns-proxy=y

CONFIG_PACKAGE_luci-app-passwall2=y
CONFIG_PACKAGE_luci-app-openclash=y

# Compress image
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_PACKAGE_luci-app-irqbalance=y
CONFIG_PACKAGE_irqbalance=y # https://github.com/4IceG/luci-app-irqbalance

VIKINGYFY, 2025-04-24:
1，pw组件不要全选，选择xray+sing-box即可，其他都是过时的
2，pw和pw2不能共存，oc可能会报rust错误
3，nss自带中断绑定，不需要这些第三方工具

intannajwa, 2025-04-24:
> 1，pw组件不要全选，选择xray+sing-box即可，其他都是过时的 2，pw和pw2不能共存，oc可能会报rust错误 3，nss自带中断绑定，不需要这些第三方工具

CONFIG_DEFAULT_TZ="Asia/Kuala_Lumpur"
CONFIG_PACKAGE_luci-app-autoreboot=y
# CONFIG_DEVEL is not set
# CONFIG_CCACHE is nit set
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_PACKAGE_kmod-dsa=y
CONFIG_PACKAGE_kmod-fuse=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-mtd-rw=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-dwc3=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb-xhci=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-eem=y
# CONFIG_PACKAGE_kmod-usb-net-cdc-ncm is not set
CONFIG_PACKAGE_kmod-usb-net-cdc-subset=y

# CONFIG_PACKAGE_kmod-usb-net-rtl8150 is not set
# CONFIG_PACKAGE_kmod-usb-net-rtl8152 is not set
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-extras=y
CONFIG_PACKAGE_kmod-usb-storage-uas=y
CONFIG_PACKAGE_kmod-nft-core=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-nft-fib=y
CONFIG_PACKAGE_kmod-nft-fullcone=y
CONFIG_PACKAGE_kmod-nft-nat=y
CONFIG_PACKAGE_kmod-nft-netdev=y
CONFIG_PACKAGE_kmod-nft-offload=y
CONFIG_PACKAGE_kmod-nft-queue=y
CONFIG_PACKAGE_kmod-nft-socket=y
CONFIG_PACKAGE_kmod-nft-tproxy=y
CONFIG_PACKAGE_kmod-fs-btrfs=y
CONFIG_PACKAGE_kmod-inet-diag=y
CONFIG_PACKAGE_kmod-netlink-diag=y
CONFIG_PACKAGE_kmod-sound-core=y
CONFIG_PACKAGE_autocore=y
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_cfdisk=y
CONFIG_PACKAGE_cgdisk=y
# CONFIG_PACKAGE_coremark is not set
CONFIG_PACKAGE_cpufreq=y
# CONFIG_PACKAGE_fdisk is not set
CONFIG_PACKAGE_gdisk=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_sfdisk=y
CONFIG_PACKAGE_sgdisk=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ipkg=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-proto-bonding=y
CONFIG_PACKAGE_luci-proto-relay=y

# Quectel Cellular - Adding support
CONFIG_PACKAGE_quectel-CM-5G=y     

# Quectel-specific USB support
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan-quectel=y 
CONFIG_PACKAGE_kmod-usb-net-cdc-mbim=y
# CONFIG_PACKAGE_kmod-usb-net-cdc-ether is not set
CONFIG_PACKAGE_kmod-usb-net-rndis=y 
# CONFIG_PACKAGE_kmod-usb-net-ipheth is not set
# CONFIG_PACKAGE_kmod-usb-net-huawei-cdc-ncm is not set
CONFIG_PACKAGE_chat=y
# CONFIG_PACKAGE_comgt is not set
# CONFIG_PACKAGE_comgt-ncm is not set
CONFIG_PACKAGE_comgt-directip=y
CONFIG_PACKAGE_luci-proto-qmi=y
# CONFIG_PACKAGE_luci-proto-ncm is not set
# CONFIG_PACKAGE_luci-proto-mbim is not set
# CONFIG_PACKAGE_umbim is not set
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_usbutils=y
# CONFIG_PACKAGE_minicom is not set
# CONFIG_PACKAGE_screen is not set
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_traceroute=y
CONFIG_PACKAGE_bind-dig=y
CONFIG_PACKAGE_ping=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-turboacc=y
CONFIG_PACKAGE_TURBOACC_INCLUDE_FLOW_OFFLOADING=y
CONFIG_PACKAGE_TURBOACC_INCLUDE_BBR_CCA=y

# Update Quectel-specific packages
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-ipv6=y

# === Custom Packages from ipq-packages ===
CONFIG_PACKAGE_luci-app-3ginfo-lite=y
CONFIG_PACKAGE_luci-app-atcommands=y
CONFIG_PACKAGE_luci-app-modemband=y
CONFIG_PACKAGE_luci-app-sms-tool-js=y
CONFIG_PACKAGE_modemband=y
CONFIG_PACKAGE_sms-tool=y

# Passwall
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_luci-app-passwall2=y
CONFIG_PACKAGE_luci-app-openclash=y

# Compress image
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_PACKAGE_luci-app-irqbalance=y
CONFIG_PACKAGE_irqbalance=y
 
OK?

intannajwa, 2025-04-24:
> 1，pw组件不要全选，选择xray+sing-box即可，其他都是过时的 2，pw和pw2不能共存，oc可能会报rust错误 3，nss自带中断绑定，不需要这些第三方工具

```CONFIG_DEFAULT_TZ="Asia/Kuala_Lumpur"
CONFIG_PACKAGE_luci-app-autoreboot=y
# CONFIG_DEVEL is not set
# CONFIG_CCACHE is nit set
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_PACKAGE_kmod-dsa=y
CONFIG_PACKAGE_kmod-fuse=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-mtd-rw=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-dwc3=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb-xhci=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-eem=y
# CONFIG_PACKAGE_kmod-usb-net-cdc-ncm is not set
CONFIG_PACKAGE_kmod-usb-net-cdc-subset=y

# CONFIG_PACKAGE_kmod-usb-net-rtl8150 is not set
# CONFIG_PACKAGE_kmod-usb-net-rtl8152 is not set
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-extras=y
CONFIG_PACKAGE_kmod-usb-storage-uas=y
CONFIG_PACKAGE_kmod-nft-core=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-nft-fib=y
CONFIG_PACKAGE_kmod-nft-fullcone=y
CONFIG_PACKAGE_kmod-nft-nat=y
CONFIG_PACKAGE_kmod-nft-netdev=y
CONFIG_PACKAGE_kmod-nft-offload=y
CONFIG_PACKAGE_kmod-nft-queue=y
CONFIG_PACKAGE_kmod-nft-socket=y
CONFIG_PACKAGE_kmod-nft-tproxy=y
CONFIG_PACKAGE_kmod-fs-btrfs=y
CONFIG_PACKAGE_kmod-inet-diag=y
CONFIG_PACKAGE_kmod-netlink-diag=y
CONFIG_PACKAGE_kmod-sound-core=y
CONFIG_PACKAGE_autocore=y
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_cfdisk=y
CONFIG_PACKAGE_cgdisk=y
# CONFIG_PACKAGE_coremark is not set
CONFIG_PACKAGE_cpufreq=y
# CONFIG_PACKAGE_fdisk is not set
CONFIG_PACKAGE_gdisk=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_sfdisk=y
CONFIG_PACKAGE_sgdisk=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ipkg=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-proto-bonding=y
CONFIG_PACKAGE_luci-proto-relay=y

# Quectel Cellular - Adding support
CONFIG_PACKAGE_quectel-CM-5G=y     

# Quectel-specific USB support
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan-quectel=y 
CONFIG_PACKAGE_kmod-usb-net-cdc-mbim=y
# CONFIG_PACKAGE_kmod-usb-net-cdc-ether is not set
CONFIG_PACKAGE_kmod-usb-net-rndis=y 
# CONFIG_PACKAGE_kmod-usb-net-ipheth is not set
# CONFIG_PACKAGE_kmod-usb-net-huawei-cdc-ncm is not set
CONFIG_PACKAGE_chat=y
# CONFIG_PACKAGE_comgt is not set
# CONFIG_PACKAGE_comgt-ncm is not set
CONFIG_PACKAGE_comgt-directip=y
CONFIG_PACKAGE_luci-proto-qmi=y
# CONFIG_PACKAGE_luci-proto-ncm is not set
# CONFIG_PACKAGE_luci-proto-mbim is not set
# CONFIG_PACKAGE_umbim is not set
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_usbutils=y
# CONFIG_PACKAGE_minicom is not set
# CONFIG_PACKAGE_screen is not set
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_traceroute=y
CONFIG_PACKAGE_bind-dig=y
CONFIG_PACKAGE_ping=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-turboacc=y
CONFIG_PACKAGE_TURBOACC_INCLUDE_FLOW_OFFLOADING=y
CONFIG_PACKAGE_TURBOACC_INCLUDE_BBR_CCA=y

# Update Quectel-specific packages
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-ipv6=y

# === Custom Packages from ipq-packages ===
CONFIG_PACKAGE_luci-app-3ginfo-lite=y
CONFIG_PACKAGE_luci-app-atcommands=y
CONFIG_PACKAGE_luci-app-modemband=y
CONFIG_PACKAGE_luci-app-sms-tool-js=y
CONFIG_PACKAGE_modemband=y
CONFIG_PACKAGE_sms-tool=y

# Passwall
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_luci-app-passwall2=y
CONFIG_PACKAGE_luci-app-openclash=y

# Compress image
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_PACKAGE_luci-app-irqbalance=y
CONFIG_PACKAGE_irqbalance=y```

intannajwa, 2025-04-24:
Do u have quectel-cm repo?

VIKINGYFY, 2025-04-25:
没有

intannajwa, 2025-04-25:
你可以帮我生成一个 AW1000 的 .config 文件吗？我只需要 QModem、3ginfo Lite 和 ModemBand。我可以付费给你。但是每次我添加 QModem 插件时都会出错。

intannajwa, 2025-04-25:
.config

https://github.com/intannajwa/OpenWRT-CI/blob/main/Config/GENERAL.txt

intannajwa, 2025-04-25:
https://github.com/intannajwa/OpenWRT-CI/actions/runs/14655010490/job/41128426343

closed 2025-04-25 by intannajwa

VIKINGYFY, 2025-04-26:
> 你可以帮我生成一个 AW1000 的 .config 文件吗？我只需要 QModem、3ginfo Lite 和 ModemBand。我可以付费给你。但是每次我添加 QModem 插件时都会出错。

直接用
CONFIG_PACKAGE_luci-app-qmodem=y
即可，源我已经添加了

intannajwa, 2025-04-26:
PassWall 和 PassWall2 不能一起安装，不然编译固件会出错。我也不知道为什么，但在用 LEDE 的时候却能正常使用，或者是我的配置有问题。