VIKINGYFY/OpenWRT-CI #160: 红米AX5标准版刷带WIFI的一直闪黄灯

yyangv, 2025-04-25:
不带WIFI的能正常刷入，带WIFI的不行。

我看到这个Issuehttps://github.com/VIKINGYFY/OpenWRT-CI/issues/107，也是同样的问题，但是后面就解决了，原因未知。

他那个512MB内存，我这个235MB内存，感觉更容易出问题，您指导一下这个无线咋修？

我刷的暗云的Uboot，不知道有没有帮助。

VIKINGYFY, 2025-04-25:
256内存没法用带wifi固件，内存太小了，开不了机，64位系统内存占用很高

yyangv, 2025-04-25:
> 256内存没法用带wifi固件，内存太小了，开不了机，64位系统内存占用很高

您还记得红米ax3000那个issue吗？https://github.com/hzyitc/openwrt-redmi-ax3000/issues/84 ，他也是256内存的。是不是就意味着适配最新nss也没有希望了？

您在这个领域比较资深，打听一下ax5有没有推荐的固件？带WiFi的。

VIKINGYFY, 2025-04-25:
> > 256内存没法用带wifi固件，内存太小了，开不了机，64位系统内存占用很高
> 
> 红米ax3000有个issue [hzyitc/openwrt-redmi-ax3000#84](https://github.com/hzyitc/openwrt-redmi-ax3000/issues/84) ，他也是256内存的。是不是就意味着适配最新nss也没有希望了？
> 
> 您在这个领域比较资深，打听一下ax5有没有推荐的固件？带WiFi的。

建议改个1g，畅玩，256即使能开机，所剩内存也开不了任何插件

yyangv, 2025-04-26:
> > > 256内存没法用带wifi固件，内存太小了，开不了机，64位系统内存占用很高
> > 
> > 
> > 红米ax3000有个issue [hzyitc/openwrt-redmi-ax3000#84](https://github.com/hzyitc/openwrt-redmi-ax3000/issues/84) ，他也是256内存的。是不是就意味着适配最新nss也没有希望了？
> > 您在这个领域比较资深，打听一下ax5有没有推荐的固件？带WiFi的。
> 
> 建议改个1g，畅玩，256即使能开机，所剩内存也开不了任何插件

谢谢

closed 2025-04-26 by yyangv