VIKINGYFY/OpenWRT-CI #175: CI build报错了

zl810881283, 2025-05-31:
```
time: target/linux/install#641.85#40.39#172.99
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/install] Error 1
make[2]: Leaving directory '/mnt/build_wrt'
make[1]: *** [target/Makefile:26: /mnt/build_wrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/mnt/build_wrt'
make: *** [/mnt/build_wrt/include/toplevel.mk:233: world] Error 2
```

VIKINGYFY, 2025-06-01:
你自己看一下这段日志，能看出来什么不？

zl810881283, 2025-06-01:
已经收到您的来信，我会尽快回复。

zl810881283, 2025-06-01:
看不出来，所以来问大佬，完整日志如下

https://github.com/zl810881283/OpenWRT-CI/actions/runs/15364194467/job/43237154104#step:13:582

仅仅package根据自己需要调整了下，详细见
https://github.com/zl810881283/OpenWRT-CI/commit/0236570a1ef32eb3275596e685133ea8699f97d2

麻烦大佬帮忙看看是不是什么改错了

VIKINGYFY, 2025-06-01:
ERROR: package/feeds/packages/rust [host] failed to build.

rust相关的组件去掉

zl810881283, 2025-06-01:
我自定义的包里应该是没有rust相关组件的，全局搜了下也没找到rust相关的代码，应该是某个 luci-app 依赖rust，请教下大佬，该如何排查是哪个 luci-app的问题那？

zl810881283, 2025-06-01:
看起来是内核相关的？

[image]

VIKINGYFY, 2025-06-01:
生成一份配置文件，然后把rust相关的组件去掉

zl810881283, 2025-06-01:
发现问题所在了，只要在 GENERAL.txt 中添加 CONFIG_PACKAGE_luci-app-openclash=y 就会报这个错误，去掉后就没问题了。大佬能修复下或者给点建议吗？

VIKINGYFY, 2025-06-01:
修复不了，又不是我的问题，换个工具用

closed 2025-06-01 by VIKINGYFY