VIKINGYFY/OpenWRT-CI #176: 大佬我编译总是报这个错是什么原因

frzb2023, 2025-06-02:
/mnt/build_wrt/staging_dir/host/bin/install: cannot stat './rltest2.c': No such file or directory
make[5]: Leaving directory '/mnt/build_wrt/build_dir/hostpkg/readline-8.2/examples'
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 /mnt/build_wrt/build_dir/hostpkg/readline-8.2/readline.pc /mnt/build_wrt/staging_dir/hostpkg/lib/pkgconfig/readline.pc
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 /mnt/build_wrt/build_dir/hostpkg/readline-8.2/history.pc /mnt/build_wrt/staging_dir/hostpkg/lib/pkgconfig/history.pc
mv /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.old
mv: cannot stat '/mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a': No such file or directory
make[4]: [Makefile:255: install-static] Error 1 (ignored)
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 libreadline.a /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a
test -n "ranlib" && ranlib /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a
mv /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.old
mv: cannot stat '/mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a': No such file or directory
make[4]: [Makefile:258: install-static] Error 1 (ignored)
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 libhistory.a /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a
test -n "ranlib" && ranlib /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a
make[4]: Leaving directory '/mnt/build_wrt/build_dir/hostpkg/readline-8.2'
mkdir -p /mnt/build_wrt/staging_dir/hostpkg/stamp
touch /mnt/build_wrt/build_dir/hostpkg/readline-8.2/.built
touch /mnt/build_wrt/staging_dir/hostpkg/stamp/.readline_installed 
make[3]: Leaving directory '/mnt/build_wrt/package/libs/readline'
time: package/libs/readline/host-compile#9.51#4.12#13.87
make[2]: Leaving directory '/mnt/build_wrt'
make[1]: *** [package/Makefile:183: /mnt/build_wrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/build_wrt'
make: *** [/mnt/build_wrt/include/toplevel.mk:233: world] Error 2
Error: Process completed with exit code 2.

大佬我编译总是报这个错是什么原因

frzb2023, 2025-06-02:
Run cd ./wrt/
  cd ./wrt/
  
  make -j$(nproc) || make -j$(nproc) V=s
  shell: /usr/bin/bash -e {0}
  env:
    GITHUB_TOKEN: ***
    WRT_CONFIG: IPQ60XX-WIFI-YES
    WRT_THEME: argon
    WRT_NAME: JaWRT
    WRT_SSID: TH
    WRT_WORD: 1[2](https://github.com/frzb2023/OpenWRT-CI/actions/runs/15384046737/job/43279365562#step:13:2)345678
    WRT_IP: 192.168.[3](https://github.com/frzb2023/OpenWRT-CI/actions/runs/15384046737/job/43279365562#step:13:3)0.1
    WRT_PW: 无
    WRT_REPO: https://github.com/VIKINGYFY/immortalwrt.git
    WRT_BRANCH: main
    WRT_SOURCE: VIKINGYFY/immortalwrt
    WRT_PACKAGE: 
    WRT_TEST: 
    WRT_DATE: 25.06.02-12.51.[4](https://github.com/frzb2023/OpenWRT-CI/actions/runs/15384046737/job/43279365562#step:13:4)6
    WRT_MARK: frzb2023
    WRT_INFO: VIKINGYFY
    WRT_TARGET: qualcommax
    WRT_KVER: none
    WRT_LIST: none
    WRT_HASH: d20[5](https://github.com/frzb2023/OpenWRT-CI/actions/runs/15384046737/job/43279365562#step:13:5)ca7
WARNING: Makefile 'package/QModem/luci/luci-app-qmodem/Makefile' has a dependency on 'quectel-CM-5G', which does not exist
WARNING: Makefile 'package/feeds/packages/mqttled/Makefile' has a dependency on 'python3-netifaces', which does not exist
make[2]: Entering directory '/mnt/build_wrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/mnt/build_wrt/scripts/config'
 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/libdeflate compile
 make[3] -C tools/flock compile
 make[3] -C tools/zstd compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/tar compile
 make[3] -C tools/ninja compile
 make[3] -C tools/expat compile
 make[3] -C tools/xz compile
 make[3] -C tools/zlib compile
 make[3] -C tools/meson compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/libressl compile
 make[3] -C tools/cmake compile
 make[3] -C tools/xxhash compile
 make[3] -C tools/ccache compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/gnulib compile
 make[3] -C tools/cpio compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/lzma compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/squashfs4 compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/zip compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/libtool compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/flex compile
 make[3] -C tools/elfutils compile
 make[3] -C tools/fakeroot compile
[... 519 lines cut ...]
/mnt/build_wrt/staging_dir/host/bin/install: cannot stat './rltest2.c': No such file or directory
make[5]: Leaving directory '/mnt/build_wrt/build_dir/hostpkg/readline-8.2/examples'
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 /mnt/build_wrt/build_dir/hostpkg/readline-8.2/readline.pc /mnt/build_wrt/staging_dir/hostpkg/lib/pkgconfig/readline.pc
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 /mnt/build_wrt/build_dir/hostpkg/readline-8.2/history.pc /mnt/build_wrt/staging_dir/hostpkg/lib/pkgconfig/history.pc
mv /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.old
mv: cannot stat '/mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a': No such file or directory
make[4]: [Makefile:255: install-static] Error 1 (ignored)
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 libreadline.a /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a
test -n "ranlib" && ranlib /mnt/build_wrt/staging_dir/hostpkg/lib/libreadline.a
mv /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.old
mv: cannot stat '/mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a': No such file or directory
make[4]: [Makefile:258: install-static] Error 1 (ignored)
/mnt/build_wrt/staging_dir/host/bin/install -c -m 644 libhistory.a /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a
test -n "ranlib" && ranlib /mnt/build_wrt/staging_dir/hostpkg/lib/libhistory.a
make[4]: Leaving directory '/mnt/build_wrt/build_dir/hostpkg/readline-8.2'
mkdir -p /mnt/build_wrt/staging_dir/hostpkg/stamp
touch /mnt/build_wrt/build_dir/hostpkg/readline-8.2/.built
touch /mnt/build_wrt/staging_dir/hostpkg/stamp/.readline_installed 
make[3]: Leaving directory '/mnt/build_wrt/package/libs/readline'
time: package/libs/readline/host-compile#9.51#4.[12](https://github.com/frzb2023/OpenWRT-CI/actions/runs/15384046737/job/43279365562#step:13:12)#13.87
make[2]: Leaving directory '/mnt/build_wrt'
make[1]: *** [package/Makefile:183: /mnt/build_wrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/build_wrt'
make: *** [/mnt/build_wrt/include/toplevel.mk:[23](https://github.com/frzb2023/OpenWRT-CI/actions/runs/15384046737/job/43279365562#step:13:23)3: world] Error 2
Error: Process completed with exit code 2.

VIKINGYFY, 2025-06-02:
rust相关的报错刚修复，同步一下重新编译

frzb2023, 2025-06-03:
OK 成功了