VIKINGYFY/OpenWRT-CI #182: RPCError

Yon8, 2025-06-20:
RPCError
RPC call to luci/getFeatures failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (http://192.168.10.1/luci-static/resources/rpc.js?v=25.170.50483~50fe942:11:3)
网页报错 清除配置 依旧 自编译也依旧

Yon8, 2025-06-20:
[image]
cpu使用率不显示 温度undefined

VIKINGYFY, 2025-06-20:
刷我编译好的测试一下

cat0186, 2025-06-20:
[image]

qqsir-dev, 2025-06-20:
[image]

我复刻的也是

VIKINGYFY, 2025-06-21:
那应该是上游源码有问题，周日晚上回去看一下

qqsir-dev, 2025-06-21:
今天编译的自己又好了，P事没有了

VIKINGYFY, 2025-06-21:
应该是上游的luci或者package更新后炸了

closed 2025-06-21 by VIKINGYFY