VIKINGYFY/OpenWRT-CI #187: 请问下，最近的固件为什么自编译失败？能否解决？

RS2Tale, 2025-07-29:
感谢作者的阅读，在机缘巧合下，我一直使用的是您编译的固件，非常好用，但是最近想更新最新固件时，发现已经有两三天未能释放新的版本，我也查看了日志，好像是自编译失败，具体情况不知，想问下能否解决？我是红米AX6，想用下您前几天编辑出来的固件版本，内核应该是6.12.40

RS2Tale, 2025-07-30:
此存储库是 https://git.openwrt.org/openwrt/openwrt.git 的镜像它仅供参考，不用于签入。我们将继续在此处接受拉取请求。它们将通过暂存树合并，然后合并到 openwrt.git 中。
我看好像是不是那边路径调整了，现在是否是通过镜像授权拉取请求？具体我也不太清楚，目前还有留存备份的最新版本没，能否发我一份，我是红米AX6，谢谢大佬！

.Tale
***@***.***

        

         原始邮件

发件时间：2025年7月30日 07:57
收件人：VIKINGYFY/OpenWRT-CI ***@***.***&gt;
抄送：RS2Tale ***@***.***&gt;, Author ***@***.***&gt;

上游openwrt的原因，等着就行了
 https://github.com/openwrt/openwrt/issues

Achang233, 2025-07-31:
V佬看样子也不用着急修了
我在https://github.com/Achang233/immortalwrt 添加了来自上游的三个commit 

1.https://github.com/immortalwrt/immortalwrt/commit/66916096f3804f8c424828be9ad67f7812569a16
2.https://github.com/immortalwrt/immortalwrt/commit/2242ee631daf4961503c44e12d9f3b6a883db922
3.https://github.com/immortalwrt/immortalwrt/commit/2684fe31d02e2c88ead04bca7c464e7906682c3c

最后在https://github.com/[Achang233/immortalwrt_ipq-CI 编译的时候遇到了和
https://github.com/openwrt/packages/issues/27131 一样的因为gettext版本引起的rpcsvc-proto编译错误
看样子还得等上游修复 洗洗睡吧

kolioF, 2025-08-02:
解决了吗 V佬

Achang233, 2025-08-02:
> 解决了吗 V佬

暂时没可能 上游 https://github.com/openwrt/packages/issues/27131 还在讨论这个依赖的问题
不过如果你恰好不需要samba 或者nfs文件共享啥的，使用我fork的那个源 https://github.com/Achang233/immortalwrt  应该是可以编译的
 我和 @Zyilin98 已经在 https://github.com/Achang233/immortalwrt_ipq-CI 试过了是可以用的

VIKINGYFY, 2025-08-05:
源码修好了，可以编译了

closed 2025-08-05 by VIKINGYFY

RS2Tale, 2025-08-05:
感谢大佬，我已看见，非常感谢！昨天已经刷入，感觉非常好，延迟很低

发自我的iPhone