VIKINGYFY/OpenWRT-CI #203: 请教大神，我fork你的项目，make download -j1 V=s 出现错误，请问是怎么回事

zero-dream, 2025-09-17:
请教大神，我fork你的项目，make download -j1 V=s 出现错误，请问是怎么回事
为什么我查看你的仓库下载没有问题，但是到我这里就有问题了呢
fork 时间：2025.09.17

`Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at e472690 QCA_Networking_2022.SPF_12.2: IPQ5018: Add NSS FW 12.2-156
Generating formal git archive (apply .gitattributes rules)
Packing checkout...
Hash mismatch for file nss-firmware-2025.05.01~e4726900.tar.zst: expected 3ec87f221e8905d4b6b8b3d207b7f7c4666c3bc8db7c1f06d4ae2e78f863b8f4, got 881cbf75efafe380b5adc91bfb1f68add5e29c9274eb950bb1e815c7a3622807
make[2]: *** [Makefile:167: /home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f/dl/nss-firmware-2025.05.01~e4726900.tar.zst] Error 1
make[2]: Leaving directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f/feeds/nss_packages/firmware/nss-firmware'
time: package/feeds/nss_packages/nss-firmware/download#2.97#0.50#8.03
    ERROR: package/feeds/nss_packages/nss-firmware failed to build.
make[1]: *** [package/Makefile:183: package/feeds/nss_packages/nss-firmware/download] Error 1
make[1]: Leaving directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f'
make[1]: Entering directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f'
make[2]: Entering directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f/target/linux'
make[3]: Entering directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f/target/linux/qualcommax'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f/target/linux/qualcommax'
make[2]: Leaving directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f/target/linux'
time: target/linux/download#0.03#0.05#0.06
make[1]: Leaving directory '/home/runner/work/_temp/ZeroDreamTemp-a628094cebdf48df9fb1cb1b3ee219db/ZeroWrt-4feabc3e78a542c2b181b81415e3895f'
Error: Process completed with exit code 2.`

咨询了一下AI，给出答案：这个错误的核心是 nss-firmware软件包的下载文件哈希值与OpenWrt编译系统中预期的哈希值不匹配：

预期哈希值 (expected): 3ec87f221e8905d4b6b8b3d207b7f7c4666c3bc8db7c1f06d4ae2e78f863b8f4

实际下载文件哈希值 (got): 881cbf75efafe380b5adc91bfb1f68add5e29c9274eb950bb1e815c7a3622807

导致这种不一致的常见原因包括：

上游源更新但编译脚本未同步：软件源（如Git仓库）上的文件可能被更新了（例如，开发者提交了新的更改），但OpenWrt编译脚本（如Makefile）中对应的哈希值（PKG_HASH）没有及时更新。

网络下载问题：网络传输过程中文件损坏或下载不完全，虽然概率较低，但也是可能性之一。

版本或分支切换：如果你切换了OpenWrt或其feed的版本分支，可能导致不同分支预期的文件哈希不同。

--------------------

为什么会在我fork的仓库出现问题呢，而你的仓库没有这个问题？

zero-dream, 2025-09-17:
上面出现问题，可能是因为经过修改导致。刚刚，我直接fork ，仅修改一下内容
      - name: Download Packages
        if: env.WRT_TEST != 'true'
        run: |
          cd ./wrt/

          make download -j1
          exit 1
---
提示错误：
 make[2] -C feeds/nss_packages/firmware/nss-firmware download
    ERROR: package/feeds/nss_packages/nss-firmware failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download
Error: Process completed with exit code 1.
---
这个错误和上面是一样的错误，为什么你的仓库编译没有任何问题，但是fork的仓库会有这个问题呢？

zero-dream, 2025-09-18:
> 没法解决，下载失败会自动重试，继续编译即可

我测试了3次，每次编译都是这个错误提示，我看你仓库的编译输出就没有这个问题，这太奇怪了吧

zero-dream, 2025-09-18:
> 建议云编译，问题少，我之前也是电脑编译，也是问题多。现在已经爱上云编译了，感谢[VIKINGYFY](https://github.com/VIKINGYFY)

我这个就是fork后云编译的 0.0

hicasper, 2025-09-19:
是因为有时候下载回来的NSS源码hash不对，但是有时候又是正常的，临时解决方案就是修改WRT-CORE.yml文件
```
      - name: Download Packages
        if: env.WRT_TEST != 'true'
        run: |
          cd ./wrt/
          sed -i 's|3ec87f221e8905d4b6b8b3d207b7f7c4666c3bc8db7c1f06d4ae2e78f863b8f4|881cbf75efafe380b5adc91bfb1f68add5e29c9274eb950bb1e815c7a3622807|g' feeds/nss_packages/firmware/nss-firmware/Makefile
          make download -j$(nproc)
```

zero-dream, 2025-09-19:
> 是因为有时候下载回来的NSS源码hash不对，但是有时候又是正常的，临时解决方案就是修改WRT-CORE.yml文件
>
> ```
>       - name: Download Packages
>         if: env.WRT_TEST != 'true'
>         run: |
>           cd ./wrt/
>           sed -i [...]

这样修改岂不是通过下载校验，后面不会再次下载了吧，如果真的是下载问题，这样不就编译有问题的包了吗？

zero-dream, 2025-09-19:
> 没法解决，下载失败会自动重试，继续编译即可

你看我仓库，https://github.com/zero-dream/OpenWRT-CI/actions/workflows/QCA-ALL.yml
连续3次都是这个问题，下载失败，不可能每次的 hash 都是一样的吧

hicasper, 2025-09-20:
> 修改岂不是通过下载校验，后面就不会再次下载了，如果下载有损坏的问题，这样不就编译有问题的包

nss-firmware都是从git仓库clone回来后再次打包的，而且这只是临时解决方案，想要正常编译目前只能这么改，没有必要操心多余的事情

zero-dream, 2025-09-20:
> > 修改岂不是通过下载校验，后面就不会再次下载了，如果下载有损坏的问题，这样不就编译有问题的包
> 
> nss-firmware都是从git仓库clone回来后再次打包的，而且这只是临时解决方案，想要正常编译目前只能这么改，没有必要操心多余的事情

我查看这个作者的action日志，为什么没有这个问题呢，fork的仓库编译就有问题，都是同样的代码编译，很奇怪

hicasper, 2025-09-20:
> > > 修改岂不是通过下载校验，后面就不会再次下载了，如果下载有损坏的问题，这样不就编译有问题的包
> > 
> > 
> > nss-firmware都是从git仓库clone回来后再次打包的，而且这只是临时解决方案，想要正常编译目前只能这么改，没有必要操心多余的事情
> 
> 我查看这个作者的action日志，为什么没有这个问题呢，fork的仓库编译就有问题，都是同样的代码编译，很奇怪

你没发现最新的高通构建也失败了吗，他的构建每次都会把以前的记录清除而已。

zero-dream, 2025-09-20:
> > > > 修改岂不是通过下载校验，后面就不会再次下载了，如果下载有损坏的问题，这样不就编译有问题的包
> > > 
> > > 
> > > nss-firmware都是从git仓库clone回来后再次打包的，而且这只是临时解决方案，想要正常编译目前只能这么改，没有必要操心多余的事情
> > 
> > 
> > 我查看这个作者的action日志，为什么没有这个问题呢，fork的仓库编译就有问题，都是同样的代码编译，很奇怪
> 
> 你没发现最新的高通构建也失败了吗，他的构建每次都会把以前的记录清除而已。

他最近失败是因为最近更新的源码有问题，这是另外一个问题，昨天才出现的，我这个问题是3天前就有的

closed 2025-09-20 by zero-dream